eclipse-paho/paho.mqtt.embedded-c #13: Waspmote embedded C version

jpwsutton, 2016-02-04:
migrated from Bugzilla [#462680](https://bugs.eclipse.org/bugs/show_bug.cgi?id=462680)
status ASSIGNED severity *normal* in component *MQTT-Embedded-C* for *1.1*
Reported in version *unspecified* on platform *Other*
Assigned to: Ian Craggs

On 2015-03-20 09:00:16 -0400, Guy D wrote:

> Similar to the Arduino Paho client is it possible building one for Libelium Waspmote?
> Thanks
> Guy

On 2015-03-31 08:32:13 -0400, Ian Craggs wrote:

> Hi Guy.
> 
> so according to the website, the Waspmote has these wireless interfaces:
> 
>     802.15.4 / ZigBee
>     LoRa 868 / 915MHz
>     Bluetooth Low Energy (BLE) 4.0
>     WiFi
>     6LoWPAN / IPv6 Radio
>     3G + GPS
>     GPRS + GPS
>     GSM / GPRS
>     Bluetooh PRO
>     RFID/NFC
>     Expansion Radio Board
>     Waspmote Gateway
> 
> I think we're talking about the WiFi interface in this case, right?

On 2015-03-31 10:11:55 -0400, Guy D wrote:

> Hi Ian,
> 
> Indeed these are the available (wireless) communication modules.
> 
> Wifi is one but I don't know if it's difficult also to have it for GSM/GPRS, 3G?
> The latter two would be interesting in cases of having edge devices somewhere in the field with GSM/GPRS or 3G communication communicating with an MQTT broker in the cloud.
> 
> Thanks.
> Guy

On 2015-04-24 13:22:05 -0400, Guy D wrote:

> Hi Ian,
> 
> I don't know if you already had the chance starting?
> The most important would be GSM/GPRS and WiFi.
> 
> If you are too busy with other things, maybe you can give me some starting points so I can try to do the port myself?
> 
> Thanks.
> Guy

On 2015-04-28 09:08:50 -0400, Ian Craggs wrote:

> There are two classes that need to be written and tested to make the API work on the WaspMote.
> 
> 1) A timer class.  This is the Arduino version:
> 
> http://git.eclipse.org/c/paho/org.eclipse.paho.mqtt.embedded-c.git/tree/MQTTClient/src/arduino/Countdown.h
> 
> which relies on the Arduino millis() function.  If the WaspMote supports the millis() function too, then this class should work as is.
> 
> 2) A network class, for each networking interface you want to use.  An example for Arduino is here:
> 
> http://git.eclipse.org/c/paho/org.eclipse.paho.mqtt.embedded-c.git/tree/MQTTClient/src/arduino/WifiIPStack.h
> 
> The absolutely necessary methods are:
> 
>     int read(char* buffer, int len, int timeout);  // read data, length len, from the network into the supplied buffer.  The timeout is in milliseconds.
>     
>     int write(char* buffer, int len, int timeout); // write date length len from the buffer to the network.  Again the timeout is in milliseconds.
> 
> All the other methods are optional, for convenience.
> 
> Looking at the WaspMote wifi networking guide here: http://www.libelium.com/development/waspmote/documentation/wifi-networking-guide/, I see that Chapter 13 has a coding example.  
> 
> So the read and write methods of the networking class would have to call 
> 
> WIFI.send(buffer, len);
> 
> and 
> 
> WIFI.read(UNBLO);   // times out after 2 seconds, according to the doc
> 
> respectively, after having setup the WIFI connection.  Does this make sense?  Does it get you started?

gdillen, 2016-02-05:
Hi Ian,

I haven't followed the thread since last year, sorry for that. But will pick it up again.

Thanks.
Guy

icraggs, 2018-03-05:
I'm going to close this one for now.

closed 2018-03-05 by icraggs