eclipse-paho/paho.mqtt.embedded-c #17: Embedded C++ client cannot reconnect and resubscribe after broker failure

jpwsutton, 2016-02-04:
migrated from Bugzilla [#464169](https://bugs.eclipse.org/bugs/show_bug.cgi?id=464169)
status RESOLVED severity *normal* in component *MQTT-Embedded-C* for *1.1*
Reported in version *unspecified* on platform *Other*
Assigned to: Ian Craggs

On 2015-04-08 10:38:53 -0400, Joe Planisky wrote:

> If a client is disconnected from the broker (e.g. because the broker crashed), the same client instance can reconnect to the broker, but cannot resubscribe to the same topics.  
> 
> When a Client subscribes to a topic, an entry is added to the messageHandlers array.  If all the slots in that array are used, attempts to subscribe to additional topics fail.  There is no way to remove entries from that array once the connection to the broker is lost, so there's no way to reconnect and resubscribe once the broker comes back up without creating a new instance of the Client.
> 
> To put it another way, here's what I expect to be able to do:
> 
> - create a Client
> - connect to broker (clean_session = true)
> - subscribe to topics
> - receive pubs
> - broker crashes | network failure, Client senses problem and isconnected = false.
> - broker restarts
> - same client instance connects to broker (clean_session = true)
> - subscribe to (same) topics
> - continue receiving pubs

On 2015-04-08 16:00:52 -0400, Ian Craggs wrote:

> I've created a fix in the (newly created) develop branch, if you'd like to try it, Joe.
> 
> The second part will be to remove the message handler on unsubscribe.

On 2015-04-08 18:21:22 -0400, Joe Planisky wrote:

> Initial testing looks good, Ian.  We're able to reconnect & resubscribe after a broker crash.

On 2015-04-09 09:18:08 -0400, Ian Craggs wrote:

> Code to remove message handler on unsubscribe now also added to the develop branch.

closed 2016-02-04 by jpwsutton

w3hacker, 2016-10-16:
How to deal with this problem?? 

I checked out develop branch and run make, it gave me an error.
How to build it for arduino ?

jpwsutton, 2016-10-17:
@w3hacker This Issue is not relevant to your question and has been closed. You should either raise a new issue, Post a question the the [Eclipse Paho forum](https://www.eclipse.org/forums/index.php/f/247/) , or send it into the [Paho mailing list](https://dev.eclipse.org/mailman/listinfo/paho-dev).
You could also just download the pre-built Arduino library: https://www.eclipse.org/paho/clients/c/embedded/