eclipse-paho/paho.mqtt.embedded-c #2: MQTTDeserialize_publish uses wrong type for payloadlen parameter

jpwsutton, 2016-02-04:
migrated from Bugzilla [#481975](https://bugs.eclipse.org/bugs/show_bug.cgi?id=481975)
status UNCONFIRMED severity *critical* in component *MQTT-Embedded-C* for *1.2*
Reported in version *unspecified* on platform *PC*
Assigned to: Ian Craggs

On 2015-11-11 20:05:54 -0500, Joe McIlvain wrote:

> `MQTTDeserialize_publish` takes the `payloadlen` parameter as an `int *` (the address of an `int`), but the `MQTTMessage` struct defines `payloadlen` as of type `size_t`.
> 
> On platforms where `size_t` is bigger than `int`, this causes the upper byte(s) of the `size_t` bytes to be uninitialized/undefined. If the upper byte(s) happen to be zero, this works okay - if they happen to be anything else, the `payloadlen` number becomes huge, and will likely cause a segmentation fault when you try to read the message.
> 
> `MQTTDeserialize_publish` should be fixed to use type `size_t` for the `payloadlen` parameter.

jemc, 2016-02-09:
Thanks for migrating.

I'm the original reporter of this bug, and I'm here if you need to ask me any questions.

jemc, 2016-03-17:
How can I make forward progress with this issue ticket?

icraggs, 2016-09-14:
Hello.  Thanks for reporting and commenting and sorry it's taken so long for me to get to look at this issue.  Obviously, the correct solution would be to change the types of all the size related parameters in the MQTTPacket library to size_t.   Because the library is used in a number of places I will have to check that any knock on effects are resolved too.  I'll target this for release 1.1

icraggs, 2017-07-04:
We could just zero out the payloadlen field in MQTTMessage before calling MQTTDeserialize_publish.  After all, the MQTT payload length cannot exceed 256MB, so size_t is not actually needed.

icraggs, 2017-07-07:
I've set the payloadlen field to 0 to address this for now.

closed 2017-07-07 by icraggs