eclipse-paho/paho.mqtt.embedded-c #3: linux_read in MQTTLinux.c doesn't handle recv returning 0

jpwsutton, 2016-02-04:
migrated from Bugzilla [#481962](https://bugs.eclipse.org/bugs/show_bug.cgi?id=481962)
status UNCONFIRMED severity *major* in component *MQTT-Embedded-C* for *1.2*
Reported in version *unspecified* on platform *PC*
Assigned to: Ian Craggs

On 2015-11-11 16:18:47 -0500, Joe McIlvain wrote:

> I've observed cases where `recv` in the `linux_read` function in `MQTTClient.c` returns 0, instead of -1 or a positive number of bytes.
> 
> I'm not sure how/why this happens, but I can reliably reproduce this when I create a trivial loop of `MQTTYield(); printf(".\n"); sleep(1);`. At some point, the `.`s stop printing and the execution gets stuck in an inner, unthrottled loop at 100% CPU. This is because `recv` keeps returning 0, so the loop in `linux_read` never accumulates a number of `bytes` above the target `len`, and never stops.
> 
> I am on 64-bit Linux (Fedora 22), compiled from latest source.
> 
> I would recommend treating `rc == 0 && bytes == 0` as a condition to `break` the loop in `linux_read`.

jemc, 2016-02-09:
Thanks for migrating.

I'm the original reporter of this bug, and I'm here if you need to ask me any questions.

icraggs, 2017-07-07:
Thanks.  I've fixed this in the recent updates.  0 will be returned from linux_read, and the main client will handle it.

closed 2017-07-07 by icraggs