eclipse-paho/paho.mqtt.embedded-c #4: Keepalive Fix + Enhancement

jpwsutton, 2016-02-04:
migrated from Bugzilla [#481454](https://bugs.eclipse.org/bugs/show_bug.cgi?id=481454)
status UNCONFIRMED severity *enhancement* in component *MQTT-Embedded-C* for *1.2*
Reported in version *1.0* on platform *All*
Assigned to: Ian Craggs

Original attachment names and IDs:
- _[keepalive.diff](https://bugs.eclipse.org/bugs//attachment.cgi?id=257744)_ (ID 257744)

On 2015-11-04 15:58:22 -0500, John Rotach wrote:

> Created attachment 257744
> git patch for keepalive
> 
> Currently the return value from keepalive is unused.  This prevents client disconnect on dead TCP connections.  This change addresses this.
> 
> Also in this change is what we believe to be an enhancement to ping handing.  In our disconnect testing resetting the ping timer on a successful packet send (in sendPacket) prevented the client from determining a disconnect until an underlying TCP buffer error occurs.  Using our default Linux configuration this was much longer than the keepalive specified to the client.  We have made an update to only use the PINGREQ/PINGRESP mechanism.  Any feedback on this design would be much appreciated.

kartben, 2016-02-04:
@johnrotach @rotach

icraggs, 2017-07-10:
Sorry this has taken so long - I wasn't originally expecting to maintain the C high level client when Al started it as a copy of the C++ version.

I've changed the keep alive processing to use two timers, as does the C++ client.  I believe this covers all cases - I'll add some tests for the situation when I can.  The proposed fix is in the develop branch.

icraggs, 2017-07-14:
Test added, and works.

closed 2017-07-14 by icraggs