eclipse-paho/paho.mqtt.embedded-c #5: MQTTClient/src/MQTTClient.h missing + erroneous include statements

jpwsutton, 2016-02-04:
migrated from Bugzilla [#478665](https://bugs.eclipse.org/bugs/show_bug.cgi?id=478665)
status ASSIGNED severity *normal* in component *MQTT-Embedded-C* for *1.2*
Reported in version *1.0* on platform *PC*
Assigned to: Ian Craggs

On 2015-09-29 15:04:04 -0400, David H. wrote:

> MQTTClient/src/MQTTClient.h has two issues with included libraries.
> 
> 1) #include "stdio.h" should be #include <stdio.h>
> 2) #include <string.h> needs to be added for memcpy to work (called on lines 695 and 868)
> 
> I noticed the memcpy error when I ran build.sh in MQTTClient/samples/linux; output below:
> 
> me:linux:% sh build.sh
> In file included from hello.cpp:3:0:
> ../../src/MQTTClient.h: In instantiation of âint MQTT::Client<Network, Timer, MAX_MQTT_PACKET_SIZE, MAX_MESSAGE_HANDLERS>::connect(MQTTPacket_connectData&) [with Network = IPStack; Timer = Countdown; int MAX_MQTT_PACKET_SIZE = 100; int MAX_MESSAGE_HANDLERS = 5]â:
> hello.cpp:42:29:   required from here
> ../../src/MQTTClient.h:695:15: error: âmemcpyâ was not declared in this scope, and no declarations were found by argument-dependent lookup at the point of instantiation [-fpermissive]
>          memcpy(sendbuf, pubbuf, MAX_MQTT_PACKET_SIZE);
>                ^
> In file included from ../../src/linux/linux.cpp:32:0,
>                  from hello.cpp:7:
> /usr/include/string.h:46:14: note: âvoid* memcpy(void*, const void*, size_t)â declared here, later in the translation unit
>  extern void *memcpy (void *__restrict __dest, const void *__restrict __src,
>               ^
> In file included from hello.cpp:3:0:
> ../../src/MQTTClient.h: In instantiation of âint MQTT::Client<Network, Timer, MAX_MQTT_PACKET_SIZE, MAX_MESSAGE_HANDLERS>::publish(const char*, void*, size_t, short unsigned int&, MQTT::QoS, bool) [with Network = IPStack; Timer = Countdown; int MAX_MQTT_PACKET_SIZE = 100; int MAX_MESSAGE_HANDLERS = 5; size_t = long unsigned int]â:
> ../../src/MQTTClient.h:888:19:   required from âint MQTT::Client<Network, Timer, MAX_MQTT_PACKET_SIZE, MAX_MESSAGE_HANDLERS>::publish(const char*, void*, size_t, MQTT::QoS, bool) [with Network = IPStack; Timer = Countdown; int MAX_MQTT_PACKET_SIZE = 100; int MAX_MESSAGE_HANDLERS = 5; size_t = long unsigned int]â
> ../../src/MQTTClient.h:895:19:   required from âint MQTT::Client<Network, Timer, MAX_MQTT_PACKET_SIZE, MAX_MESSAGE_HANDLERS>::publish(const char*, MQTT::Message&) [with Network = IPStack; Timer = Countdown; int MAX_MQTT_PACKET_SIZE = 100; int MAX_MESSAGE_HANDLERS = 5]â
> hello.cpp:61:39:   required from here
> ../../src/MQTTClient.h:868:15: error: âmemcpyâ was not declared in this scope, and no declarations were found by argument-dependent lookup at the point of instantiation [-fpermissive]
>          memcpy(pubbuf, sendbuf, len);
>                ^
> In file included from ../../src/linux/linux.cpp:32:0,
>                  from hello.cpp:7:
> /usr/include/string.h:46:14: note: âvoid* memcpy(void*, const void*, size_t)â declared here, later in the translation unit
>  extern void *memcpy (void *__restrict __dest, const void *__restrict __src,
>               ^
> In file included from stdoutsub.cpp:40:0:
> ../../src/MQTTClient.h: In instantiation of âint MQTT::Client<Network, Timer, MAX_MQTT_PACKET_SIZE, MAX_MESSAGE_HANDLERS>::connect(MQTTPacket_connectData&) [with Network = IPStack; Timer = Countdown; int MAX_MQTT_PACKET_SIZE = 1000; int MAX_MESSAGE_HANDLERS = 5]â:
> stdoutsub.cpp:188:26:   required from here
> ../../src/MQTTClient.h:695:15: error: âmemcpyâ was not declared in this scope, and no declarations were found by argument-dependent lookup at the point of instantiation [-fpermissive]
>          memcpy(sendbuf, pubbuf, MAX_MQTT_PACKET_SIZE);
>                ^
> In file included from ../../src/linux/linux.cpp:32:0,
>                  from stdoutsub.cpp:44:
> /usr/include/string.h:46:14: note: âvoid* memcpy(void*, const void*, size_t)â declared here, later in the translation unit
>  extern void *memcpy (void *__restrict __dest, const void *__restrict __src,
> 
> 
> 
> Side note: it's unclear if stdio.h is necessary. Removing it doesn't produce any compile errors, and a quick search didn't locate any lines that use it. But I may have missed something.

On 2015-09-30 06:01:55 -0400, Ian Craggs wrote:

> I don't get any errors when I build on my RedHat installation, but I'm happy to make the change in the include.

On 2015-09-30 07:49:32 -0400, David H. wrote:

> That's odd; I'm on Fedora. I'd expect that you'd get the same errors on Red Hat. Maybe our compilers are set up differently?

elviselle, 2016-05-20:
I got the same error message on CentOS 7.

thelonelyhick, 2016-07-20:
Got the same memcpy() error on Mint Cinammon

icraggs, 2017-07-07:
I fixed the stdio.h.  The CMake build I've created works without adding <string.h>.  Not sure why, but I'll close this for now.

closed 2017-07-07 by icraggs