eclipse-paho/paho.mqtt.embedded-c #36: A networked error on .read() will not make the client disconnected.

256dpi, 2016-04-13:
Hi again,

I found another issue while working on the arduino-mqtt library.

It looks like that errors from the `read()` method in the Network class do not traverse up in the client and set the internal flag to disconnected. This has the effect that clients, that never send any packets and only read from the connection, never change the internal "connected" flag if the underlying connection fails.

A fix would be to set the client to disconnected if any serious errors happen while reading from the underlying connection. That way, a future call to `isConnected()` would correctly return `false`.

256dpi, 2017-02-20:
Is somebody interested in looking at that?

icraggs, 2017-07-07:
I've updated the code that uses read() so that it correctly uses these return values:
-1: connection error, isConnected should then return false
0: zero bytes received within the time available, no connection error
>0: the number of bytes received, no connection error

This works on Linux, I'm not sure about the Arduino read() implementation following these requirements properly.

icraggs, 2017-07-14:
I've added a test for disconnection which works on Linux.   All that remains is to check it works on Arduino.

ppescher, 2020-03-10:
@icraggs for Arduino see #205

icraggs, 2023-11-30:
Superseded by #205

closed 2023-11-30 by icraggs