eclipse-paho/paho.mqtt.embedded-c #39: pub0sub1_nb.c does not receive published message

icraggs, 2016-05-31:
Moved from issue https://github.com/eclipse/paho.mqtt.c/issues/21 - wrong repo.

migrated from Bugzilla #476397
status UNCONFIRMED severity critical in component MQTT-C for 1.2
Reported in version unspecified on platform PC
Assigned to: Ian Craggs

On 2015-09-02 03:31:11 -0400, philippe smadja wrote:

    pub0sub1.c and pub0sub1_nb.c do not work (receive message)
    qos0pub.c published properly (I gor the message with java client.
    Environement: windows 7.
    NOTE. I updated transport.c for Windows: call to WSAStartup() to initialize socket library
    Any idea.?
    Thanks
    Philippe

scaprile, 2016-11-29:
Hello Ian,
I believe the OP perhaps did not read the code before submitting this issue.
in qos0pub.c:
`        topicString.cstring = "mytopic";`
 in pub0sub1.c:
```
        /* subscribe */
        topicString.cstring = "substopic";
```
Clearly they are different topics.
Besides, there is this note:
>  NOTE. I updated transport.c for Windows: call to WSAStartup() to initialize socket library
>  Any idea.?
> 
I would point to the OP's mods first, since these worked OK when tested and the OP does not seem to provide analysis information, just a "does not work". I would expect some attempts of debugging at least.
Perhaps he could tell **what** he did modify.
I can't test on Windoze, but I can re-test on Linux.
I guess you can close this as invalid.

icraggs, 2017-07-07:
Thanks Sergio for the comment.  I'll close it.

closed 2017-07-07 by icraggs