eclipse-paho/paho.mqtt.embedded-c #40: C client's readPacket can overflow read buffer

cpacey, 2016-06-08:
The C client's [`readPacket`](https://github.com/eclipse/paho.mqtt.embedded-c/blob/8e6d4a43276f02a056865415ccad76f425f9f936/MQTTClient-C/src/MQTTClient.c#L100) reads the message size from the incoming packet and then [reads that much data into the read buffer](https://github.com/eclipse/paho.mqtt.embedded-c/blob/8e6d4a43276f02a056865415ccad76f425f9f936/MQTTClient-C/src/MQTTClient.c#L117) (`c->readbuf`) with no regard for the read buffer's size (`c->readbuf_size`).  This means that messages larger than the read buffer's size will always cause a buffer overflow.

We're using FreeRTOS, with a read buffer that's allocated on the stack of the task that's handling Mqtt.  When this buffer overflow occurs it smashes our stack (including the `MQTTClient`) and leads to anomalous behaviour.

andrewchambers, 2016-07-03:
Yes, this library seems to have many buffer overflows and other issues, I would not recommend using it as is.

eden-wang, 2017-06-14:
I found the problem too. When read buffer is not enough, I disgard the rest data. Here is the main codes
```
/* erase useless data */
read_bytes = 0;
while (read_bytes < rem_len || expired(timer))
{
    read_bytes += c->ipstack->mqttread(c->ipstack,
            c->readbuf,
            min(c->readbuf_size, rem_len - read_bytes),
            left_ms(timer));
}
```

icraggs, 2017-07-07:
I've added a fix for this in readPacket which I believe does most of the job.  If the read buffer is very small (less than the header + length of remaining length, between 0 and 5) then that could still be a problem, but buffers of that size are not very useful anyway.

closed 2017-07-07 by icraggs

AkremToujani, 2020-04-16:
Hi Teams,
I'm using paho mqtt client with NUCLEO STM32 board,
I fixed the KeepAlive Interval to 30 min (1800s) and yield(100), but, after 9 hours the client has disconnected.

any help please
thanks a lot