eclipse-paho/paho.mqtt.embedded-c #49: MQTTClient.h - (re)subscribe to the same topic filter failed

miketran78727, 2016-09-26:
Steps to reproduce the issue (e.g TI CC3200):
````
void loop() {
	int rc = 0;
	if (!client.isConnected()) {
		rc = connect();
	}
	
	if (rc == 0) {
		char json[56];
		sprintf(json, "{\"d\":{\"count\":%d}}\0", count++);
		Serial.print("Publishing: ");
		Serial.println(json);
		MQTT::Message message;
		message.qos = MQTT::QOS0; 
		message.retained = false;
		message.payload = json; 
		message.payloadlen = strlen(json);
		rc = client.publish(pubtopic, message);
		if (rc != 0) {
			Serial.print("Message publish failed with return code : ");
			Serial.println(rc);
		}
		
		Serial.print("Subscribing to topic : ");
		Serial.print(subtopic);
		rc = client.subscribe(subtopic, MQTT::QOS1, messageArrived);
		if (rc != 0) {
			Serial.print(" Subscribe failed with return code : ");
			Serial.println(rc);
		} else {
			Serial.println(" ...OK");
			client.yield(1000);
			Serial.print("Unsubscribing from topic : ");
			Serial.println(subtopic);
			rc = client.unsubscribe(subtopic);
			if (rc != 0) {
				Serial.print("  Unsubscribe failed with return code : ");
				Serial.println(rc);
			} else {
				Serial.println(" ...OK");
			}
		}
		
		if (rc != 0) {
			client.yield(60000);
		}

	}
	
	// Wait 10 seconds before publishing again
	client.yield(10000);
}
````

After iteration MAX_MESSAGE_HANDLERS (default is 5) the subscribe() failed as shown in the console:
````
Subscribe failed with return code : 1
````
I am testing my simple fix for this issue on CC3200,..

icraggs, 2016-10-11:
Hi Mike.  Thanks, I've merged the fix for the C++ client.  I notice that the same problem exists in the C version, which is not surprising as they are based on the same code.  I'll leave this issue open until I've fixed the other code too.

rafaeldelucena, 2017-02-07:
Hello @icraggs can you review the C part of the solution at #55?

Kind regards

icraggs, 2017-05-15:
Both done now.

closed 2017-05-15 by icraggs