eclipse-paho/paho.mqtt.embedded-c #53: target 'MQTTPacket/src/../test/test1' doesn't match the target pattern

wujinghust, 2016-11-12:
I use cross compiler "arm-linux-gnueabihf-gcc" to make the makefile, but it warns as fellows:

Makefile:101: target 'MQTTPacket/src/../test/test1' doesn't match the target pattern
Makefile:105: target 'MQTTPacket/src/../samples/pub0sub1' doesn't match the target pattern
Makefile:105: target 'MQTTPacket/src/../samples/qos0pub' doesn't match the target pattern

/home/tcwg-buildslave/workspace/tcwg-make-release/label/tcwg-x86_64-ex40/target/arm-linux-gnueabihf/snapshots/glibc.git~release-2.21-master/csu/../sysdeps/arm/start.S:119: undefined reference to `main'

So, do I need to alter the Makefile?

icraggs, 2017-07-10:
I'm not sure what this error means - I've had a quick look but it wasn't obvious to me.

icraggs, 2017-07-14:
There is now a CMake build which might be better for cross compiling.

closed 2017-07-14 by icraggs