eclipse-paho/paho.mqtt.embedded-c #57: MQTTPacket_decodenb() failure mode

scaprile, 2016-11-29:
Hi everyone, hi Ian
I happened to have found a bug in my MQTTPacket.c non-blocking
functions, particularly the one that extracts the message length:
MQTTPacket_decodenb()

On really slow links, if TCP (or whatever underlying transport) fails to
provide the whole length bytes before this function is called 4 times,
next call will exit with an error code. The problem is that the function
is also counting "additional entries" in addition to properly counting
received bytes.
Correction is attached: [correctdecodenb.zip](https://github.com/eclipse/paho.mqtt.embedded-c/files/617984/correctdecodenb.zip)
Basically:

```
diff --git a/MQTTPacket/src/MQTTPacket.c b/MQTTPacket/src/MQTTPacket.c
index bd5f90a..e4746cd 100644
--- a/MQTTPacket/src/MQTTPacket.c
+++ b/MQTTPacket/src/MQTTPacket.c
@@ -333,14 +333,15 @@ static int MQTTPacket_decodenb(MQTTTransport *trp)
 	}
 	do {
 		int frc;
-		if (++(trp->len) > MAX_NO_OF_REMAINING_LENGTH_BYTES)
+		if (trp->len >= MAX_NO_OF_REMAINING_LENGTH_BYTES)
 			goto exit;
 		if ((frc=(*trp->getfn)(trp->sck, &c, 1)) == -1)
 			goto exit;
 		if (frc == 0){
 			rc = 0;
 			goto exit;
-		}
+		}
+		++(trp->len);
 		trp->rem_len += (c & 127) * trp->multiplier;
 		trp->multiplier *= 128;
 	} while ((c & 128) != 0);

````
PS: This bug hunting has been done as part of a project ending nov 30th, which will also provide new samples and a serial transport to Paho MQTT. See you later guys.

scaprile, 2016-11-29:
See also #16

icraggs, 2017-07-07:
Merged the PR.  Thanks!

PR #97 (2017-07-07): Fix #16 and #57
 Fix #16:

PINGRESP has no remaining length, but both MQTTPacket_read() and MQTTPacket_readnb() wait for some bytes after header.
MQTTPacket_read() would eventually exit when the socket read times out and returns 0 read (if the underlying transport does that), but MQTTPacket_readnb() would stay waiting for data.
This fix causes both functions to not ask for more data when remaining length is zero.

 Fix #57:

On really slow links, if TCP (or whatever underlying transport) failed to
provide the whole length bytes before this function was called 4 times,
next call would exit with [... cut]
commit ddf93f68 (2017-07-07): Fix #16:

PINGRESP has no remaining length, but both MQTTPacket_read() and MQTTPacket_readnb() wait for some bytes after header.
MQTTPacket_read() would eventually exit when the socket read times out and returns 0 read (if the underlying transport does that), but MQTTPacket_readnb() would stay waiting for data.
This fix causes both functions to not ask for more data when remaining length is zero.

 Fix #57:

On really slow links, if TCP (or whatever underlying transport) failed to
provide the whole length bytes before this function was called 4 times,
next call would exit with an error code. T [... cut]
closed 2017-07-07 by icraggs