eclipse-paho/paho.mqtt.embedded-c #64: Sequence of session present flag in MQTTConnackFlags

tcluo, 2016-12-08:
Hi all,

I have a question about the struct of MQTTConnackFlags.
According to MQTT3.1.1 specification, bit 0 of Connect Acknowledge Flags is the Session Present Flag. Does the implementation follow the specification?

typedef union
{
	unsigned char all;	/**< all connack flags */
#if defined(REVERSED)
	struct
	{
		unsigned int sessionpresent : 1;    /**< session present flag */
		unsigned int : 7;	  	          /**< unused */
	} bits;
#else
	struct
	{
		unsigned int : 7;	     			/**< unused */
		unsigned int sessionpresent : 1;    /**< session present flag */
	} bits;
#endif
} MQTTConnackFlags;	/**< connack flags byte */

Regards,
TC

monstrenyatko, 2017-01-02:
I have noticed the same problem on my Arduino Pro Mini (ATmega328).
Definition is reversed (not correct).

After simple [fix](https://github.com/monstrenyatko/ArduinoMqtt/commit/4589a1ae15f07814c3feabfbe40c55bca0943029) the `sessionpresent` flag value is returned correctly from `Mosquitto` broker.

icraggs, 2017-07-10:
Fixed added to develop branch.  Thanks.

closed 2017-07-10 by icraggs