eclipse-paho/paho.mqtt.embedded-c #73: Subscribe using PAHO Embedded C client

Forums2Mike, 2017-01-02:
I am using PAHO embedded C client for MQTT implementation. I have successfully implemented publish and now working on subscribe. I see a while loop when a SUBACK is received and the code is checking for PUBLISH value (I am referring a file pub0sub1.c from folder \\org.eclipse.paho.mqtt.embedded-c-1.0.0\MQTTPacket\samples).
It looks that the code will remain in this loop till a message is received. How to break this loop in case no message is received or desired message is received?
I tried adding a timeout (using timer) and also checking the received message but it is not working. Has anyone tried using this example to implement subscribe? Any type of help will be highly appreciated.

Thanks in advance.

Mike

icraggs, 2017-07-08:
MQTTPacket_read() and other similar functions were originally intended to be examples, helpers that would get people started on the implementation that they needed, but wouldn't necessarily do everything.  As they've become more helpful, it seems expectations have risen too.

The last parameter to MQTTPacket_read() is a callback function to read from the socket/transport layer.  That callback function has as its input the number of bytes to read.  It could/should wait on the  socket for that number of bytes until one of the following conditions is met:
1. All the expected bytes are received.  Return that byte count.
2. A timeout, if set, expires. Return the number of bytes read so far.  Could be 0.
3. An error on the socket is received.  Return -1

That's the basis on which I would start.  The higher level clients in MQTTClient and MQTTClient-C do exactly this.

icraggs, 2017-07-11:
I'm going to close this - I realize I look a long time to answer.  Please reopen if you have further questions.

closed 2017-07-11 by icraggs