eclipse-paho/paho.mqtt.embedded-c #74: subscribing topic with qos2 problem; published with that topic message is not received

hardikharpal, 2017-01-10:
Only 19-20 times message received successfully subscribed with qos2. I have tried it with multiple payload length result was same.I am running it on ubuntu.

scaprile, 2017-01-10:
A published message's QoS is the highest among the receiver's and the sender's; if you subscribe with 2 and publish with 0, it will be sent to you as 0. It should arrive anyway because it is TCP, but it would be nice from you to also describe where are you publishing to (the broker) and how are you sure that is not the problem. Unless there is a capture file showing the broker received a proper message and attempted to deliver to either the library or the client, I wouldn't bother to look at the code for possible errors. Even though, you might be having connection problems and your session be not persistent... (a clean one). Need to see the traffic.
What are you using ?... the library ?, the client ?
If you are using the library (MQTTPacket), please describe how.
If you are using the client (C or C++ ?) I can't help, wait for someone else. However, I suggest you do the Wireshark stuff.

hardikharpal, 2017-01-11:
@scaprile 
Test broker "iot.eclipse.org" port1883 
I am using MQTT C client. **topic_A/hardik** subscribed with qos2 running on ubuntu in virtual box.

And publishing message from MQTTlens   on topic  **topic_A/hardik** with qos2 also subscribed with same topic with qos2 . MQTTlens works fine with  pubsub so no u=issue on broker side.

As per your suggestion  wireshark logs are attached to this comment. My client is not sending PUBCOMP message.After looking at MQTT C client code, it does not processing PUBREL message and  not sending PUBCOMP message back to broker. `int cycle(MQTTClient* c, Timer* timer)
{
    // read the socket, see what work is due
    unsigned short packet_type = readPacket(c, timer);
    if (packet_type == 0)
        return FAILURE; // no more data to read, unrecoverable

    int len = 0,
        rc = SUCCESS;

    switch (packet_type)
    {
        case CONNACK:
        case PUBACK:
        case SUBACK:
            break;
        case PUBLISH:
        {
            MQTTString topicName;
            MQTTMessage msg;
            int intQoS;
            if (MQTTDeserialize_publish(&msg.dup, &intQoS, &msg.retained, &msg.id, &topicName,
               (unsigned char**)&msg.payload, (int*)&msg.payloadlen, c->readbuf, c->readbuf_size) != 1)
                goto exit;
            msg.qos = (enum QoS)intQoS;
            deliverMessage(c, &topicName, &msg);
            if (msg.qos != QOS0)
            {
                if (msg.qos == QOS1)
                    len = MQTTSerialize_ack(c->buf, c->buf_size, PUBACK, 0, msg.id);
                else if (msg.qos == QOS2)
                    len = MQTTSerialize_ack(c->buf, c->buf_size, PUBREC, 0, msg.id);
                if (len <= 0)
                    rc = FAILURE;
                else
                    rc = sendPacket(c, len, timer);
                if (rc == FAILURE)
                    goto exit; // there was a problem
            }
            break;
        }
        case PUBREC:
        {
            unsigned short mypacketid;
            unsigned char dup, type;
            if (MQTTDeserialize_ack(&type, &dup, &mypacketid, c->readbuf, c->readbuf_size) != 1)
                rc = FAILURE;
            else if ((len = MQTTSerialize_ack(c->buf, c->buf_size, PUBREL, 0, mypacketid)) <= 0)
                rc = FAILURE;
            else if ((rc = sendPacket(c, len, timer)) != SUCCESS) // send the PUBREL packet
                rc = FAILURE; // there was a problem
            if (rc == FAILURE)
                goto exit; // there was a problem
            break;
        }
        case PUBCOMP:
            break;
        case PINGRESP:
            c->ping_outstanding = 0;
            break;
    }
    keepalive(c);
exit:
    if (rc == SUCCESS && packet_type != FAILURE)
        rc = packet_type;
    return rc;
}`
[image]

monstrenyatko, 2017-01-11:
@hardikharpal 

In Paho MQTT C-Client `cycle` I don't see the switch-case to process PUBREL message. It can't reply with PUBCOMP.

If your build environment has c++ compiler with some C++11 support you can try to use my forked and reworked [version](https://github.com/monstrenyatko/ArduinoMqtt/blob/master/src/MqttClient.h) of Paho C++ MQTT Client (see `processPacket` method).

My project is targeted for Arduino environment but all resources including time are external so should not be a problem to use with any other environment.

RomNexus6, 2017-03-07:
same problem for me with c++.
with QOS2 it fails.

I solved making a little modification at MQTTClient.h, line 840.
from this:

#elif MQTTCLIENT_QOS2

to this:

#endif
#if MQTTCLIENT_QOS2

let me know if this solves your issue

eden-wang, 2017-05-06:
I also opened an issue for qos2 problem #89

icraggs, 2017-07-07:
I've fixed the PUBREL case, sending of PUBCOMP, and the preprocessor definitions.  For both the C and C++ clients, the Travis test using all QoSs, including QoS2, now works.  Thanks.

closed 2017-07-07 by icraggs