eclipse-paho/paho.mqtt.embedded-c #76: Looking for programmers guide and reference

nkolban, 2017-01-19:
I'd like to use this C client library in a project but I am having trouble finding documentation on how to use the APIs.  Reading the code and code samples aren't cutting it for me.   Is there any documentation on using the APIs?

PravallikaKG, 2017-06-08:
Yes, I am also looking for the same. Please provide the link if you got it

sommmen, 2018-01-17:
have you read the MQTT specs? maybe browse through that quick;

http://docs.oasis-open.org/mqtt/mqtt/v3.1.1/os/mqtt-v3.1.1-os.html#_Toc398718028

the code seems quite well commented, i found i pretty easy to understand (but then this is what im studying for).

jedidiahuang, 2018-04-09:
# Missing MQTTSerialize_purrec() API?
Dear Craggs,

I have a issue of QoS 2 when subscribing a topic. 
When I received  PUBLISH packet from Broker, I respond Broker PUBREC packet. 
I did not find MQTTSerialize_pubrec () API. So I call MQTTSerialize_ack() API to reply broker an PUBREC 
i.e.:  MQTTSerialize_ack(ls_buf, MAX_BUF_LEN, PUBREC, 0, msgid);
After receiving PUBREL, I call MQTTSerialize_purcomp() to complete QoS2 hand-shaking.
But it didn't work. After receiving PUBLISH from Broker, I reply PUBREC, but Broker did not receive PUBREC, instead it close socket after timeout.  

Please help and thanks a lot.

Best Regards,
Victor