eclipse-paho/paho.mqtt.embedded-c #82: C Client for Linux fails to compile

rafaeldelucena, 2017-02-07:
Hi folks! :)

I'm trying to build the C client for an embedded Linux distro and I see that the build.sh script fails to compile

```
rafael@backbone:~/paho.mqtt.embedded-c/MQTTClient-C/samples/linux$ ./build.sh 
In file included from stdoutsub.c:44:0:
MQTTClient.h:70:23: error: unknown type name ‘Timer’
 extern void TimerInit(Timer*);
                       ^
MQTTClient.h:71:28: error: unknown type name ‘Timer’
 extern char TimerIsExpired(Timer*);
                            ^
MQTTClient.h:72:30: error: unknown type name ‘Timer’
 extern void TimerCountdownMS(Timer*, unsigned int);
                              ^
MQTTClient.h:73:28: error: unknown type name ‘Timer’
 extern void TimerCountdown(Timer*, unsigned int);
                            ^
MQTTClient.h:74:24: error: unknown type name ‘Timer’
 extern int TimerLeftMS(Timer*);
                        ^
MQTTClient.h:114:5: error: unknown type name ‘Network’
     Network* ipstack;
     ^
MQTTClient.h:115:5: error: unknown type name ‘Timer’
     Timer ping_timer;
     ^
MQTTClient.h:132:51: error: unknown type name ‘Network’
 DLLExport void MQTTClientInit(MQTTClient* client, Network* network, unsigned int command_timeout_ms,
                                                   ^
stdoutsub.c: In function ‘usage’:
stdoutsub.c:68:2: warning: implicit declaration of function ‘exit’ [-Wimplicit-function-declaration]
  exit(-1);
  ^
stdoutsub.c:68:2: warning: incompatible implicit declaration of built-in function ‘exit’
stdoutsub.c:68:2: note: include ‘<stdlib.h>’ or provide a declaration of ‘exit’
stdoutsub.c: In function ‘getopts’:
stdoutsub.c:128:17: warning: implicit declaration of function ‘atoi’ [-Wimplicit-function-declaration]
     opts.port = atoi(argv[count]);
                 ^
stdoutsub.c: In function ‘main’:
stdoutsub.c:212:2: error: unknown type name ‘Network’
  Network n;
  ^
stdoutsub.c:213:2: error: unknown type name ‘Client’
  Client c;
  ^
stdoutsub.c:218:2: warning: implicit declaration of function ‘NewNetwork’ [-Wimplicit-function-declaration]
  NewNetwork(&n);
  ^
stdoutsub.c:219:2: warning: implicit declaration of function ‘ConnectNetwork’ [-Wimplicit-function-declaration]
  ConnectNetwork(&n, opts.host, opts.port);
  ^
stdoutsub.c:220:13: error: expected identifier or ‘(’ before ‘&’ token
  MQTTClient(&c, &n, 1000, buf, 100, readbuf, 100);
             ^
stdoutsub.c:233:19: warning: passing argument 1 of ‘MQTTConnect’ from incompatible pointer type [-Wincompatible-pointer-types]
  rc = MQTTConnect(&c, &data);
                   ^
In file included from stdoutsub.c:44:0:
MQTTClient.h:140:15: note: expected ‘MQTTClient * {aka struct MQTTClient *}’ but argument is of type ‘int *’
 DLLExport int MQTTConnect(MQTTClient* client, MQTTPacket_connectData* options);
               ^
stdoutsub.c:237:21: warning: passing argument 1 of ‘MQTTSubscribe’ from incompatible pointer type [-Wincompatible-pointer-types]
  rc = MQTTSubscribe(&c, topic, opts.qos, messageArrived);
                     ^
In file included from stdoutsub.c:44:0:
MQTTClient.h:156:15: note: expected ‘MQTTClient * {aka struct MQTTClient *}’ but argument is of type ‘int *’
 DLLExport int MQTTSubscribe(MQTTClient* client, const char* topicFilter, enum QoS, messageHandler);
               ^
stdoutsub.c:242:13: warning: passing argument 1 of ‘MQTTYield’ from incompatible pointer type [-Wincompatible-pointer-types]
   MQTTYield(&c, 1000); 
             ^
In file included from stdoutsub.c:44:0:
MQTTClient.h:176:15: note: expected ‘MQTTClient * {aka struct MQTTClient *}’ but argument is of type ‘int *’
 DLLExport int MQTTYield(MQTTClient* client, int time);
               ^
stdoutsub.c:247:17: warning: passing argument 1 of ‘MQTTDisconnect’ from incompatible pointer type [-Wincompatible-pointer-types]
  MQTTDisconnect(&c);
                 ^
In file included from stdoutsub.c:44:0:
MQTTClient.h:169:15: note: expected ‘MQTTClient * {aka struct MQTTClient *}’ but argument is of type ‘int *’
 DLLExport int MQTTDisconnect(MQTTClient* client);
               ^
stdoutsub.c:248:3: error: request for member ‘disconnect’ in something not a structure or union
  n.disconnect(&n);
   ^
In file included from MQTTClient.c:16:0:
MQTTClient.h:70:23: error: unknown type name ‘Timer’
 extern void TimerInit(Timer*);
                       ^
MQTTClient.h:71:28: error: unknown type name ‘Timer’
 extern char TimerIsExpired(Timer*);
                            ^
MQTTClient.h:72:30: error: unknown type name ‘Timer’
 extern void TimerCountdownMS(Timer*, unsigned int);
                              ^
MQTTClient.h:73:28: error: unknown type name ‘Timer’
 extern void TimerCountdown(Timer*, unsigned int);
                            ^
MQTTClient.h:74:24: error: unknown type name ‘Timer’
 extern int TimerLeftMS(Timer*);
                        ^
MQTTClient.h:114:5: error: unknown type name ‘Network’
     Network* ipstack;
     ^
MQTTClient.h:115:5: error: unknown type name ‘Timer’
     Timer ping_timer;
     ^
MQTTClient.h:132:51: error: unknown type name ‘Network’
 DLLExport void MQTTClientInit(MQTTClient* client, Network* network, unsigned int command_timeout_ms,
                                                   ^
MQTTClient.c:29:50: error: unknown type name ‘Timer’
 static int sendPacket(MQTTClient* c, int length, Timer* timer)
                                                  ^
MQTTClient.c:52:36: error: unknown type name ‘Network’
 void MQTTClientInit(MQTTClient* c, Network* network, unsigned int command_timeout_ms,
                                    ^
MQTTClient.c: In function ‘decodePacket’:
MQTTClient.c:93:24: error: request for member ‘mqttread’ in something not a structure or union
         rc = c->ipstack->mqttread(c->ipstack, &i, 1, timeout);
                        ^
MQTTClient.c: At top level:
MQTTClient.c:104:38: error: unknown type name ‘Timer’
 static int readPacket(MQTTClient* c, Timer* timer)
                                      ^
MQTTClient.c: In function ‘keepalive’:
MQTTClient.c:205:9: warning: implicit declaration of function ‘TimerIsExpired’ [-Wimplicit-function-declaration]
     if (TimerIsExpired(&c->ping_timer))
         ^
MQTTClient.c:209:13: error: unknown type name ‘Timer’
             Timer timer;
             ^
MQTTClient.c:210:13: warning: implicit declaration of function ‘TimerInit’ [-Wimplicit-function-declaration]
             TimerInit(&timer);
             ^
MQTTClient.c:211:13: warning: implicit declaration of function ‘TimerCountdownMS’ [-Wimplicit-function-declaration]
             TimerCountdownMS(&timer, 1000);
             ^
MQTTClient.c:213:34: warning: implicit declaration of function ‘sendPacket’ [-Wimplicit-function-declaration]
             if (len > 0 && (rc = sendPacket(c, len, &timer)) == SUCCESS) // send the ping packet
                                  ^
MQTTClient.c: At top level:
MQTTClient.c:223:26: error: unknown type name ‘Timer’
 int cycle(MQTTClient* c, Timer* timer)
                          ^
MQTTClient.c: In function ‘MQTTYield’:
MQTTClient.c:293:5: error: unknown type name ‘Timer’
     Timer timer;
     ^
MQTTClient.c:300:13: warning: implicit declaration of function ‘cycle’ [-Wimplicit-function-declaration]
         if (cycle(c, &timer) == FAILURE)
             ^
MQTTClient.c: In function ‘MQTTRun’:
MQTTClient.c:313:2: error: unknown type name ‘Timer’
  Timer timer;
  ^
MQTTClient.c: At top level:
MQTTClient.c:340:45: error: unknown type name ‘Timer’
 int waitfor(MQTTClient* c, int packet_type, Timer* timer)
                                             ^
MQTTClient.c: In function ‘MQTTConnect’:
MQTTClient.c:357:5: error: unknown type name ‘Timer’
     Timer connect_timer;
     ^
MQTTClient.c:375:5: warning: implicit declaration of function ‘TimerCountdown’ [-Wimplicit-function-declaration]
     TimerCountdown(&c->ping_timer, c->keepAliveInterval);
     ^
MQTTClient.c:382:9: warning: implicit declaration of function ‘waitfor’ [-Wimplicit-function-declaration]
     if (waitfor(c, CONNACK, &connect_timer) == CONNACK)
         ^
MQTTClient.c: In function ‘MQTTSubscribe’:
MQTTClient.c:409:5: error: unknown type name ‘Timer’
     Timer timer;
     ^
MQTTClient.c: In function ‘MQTTUnsubscribe’:
MQTTClient.c:464:5: error: unknown type name ‘Timer’
     Timer timer;    
     ^
MQTTClient.c: In function ‘MQTTPublish’:
MQTTClient.c:503:5: error: unknown type name ‘Timer’
     Timer timer;   
     ^
MQTTClient.c: In function ‘MQTTDisconnect’:
MQTTClient.c:563:5: error: unknown type name ‘Timer’
     Timer timer;     // we might wait for incomplete incoming publishes to complete
```

I'm doing it at the wrong way or this must be fixed?

Thanks

icraggs, 2017-05-15:
Well, weird - it works for me.  I've also just added CMake builds to build a shared library and the sample.  These are invoked (and work) in Travis-CI.  Maybe those will help you figure it out?

rafaeldelucena, 2017-05-16:
I rebase with origin/master and it works! :)

icraggs, 2017-07-07:
Ok, great.  I'm also creating proper CMake build & tests along with Travis CI integration, so it should all be in better shape soon.

closed 2017-07-07 by icraggs