eclipse-paho/paho.mqtt.embedded-c #91: yield() method always return -1

miketran78727, 2017-05-17:
According to the function prolog, the yield() method should return success (0) or failure (<0)
https://github.com/eclipse/paho.mqtt.embedded-c/blob/develop/MQTTClient/src/MQTTClient.h#L187
In my tests, I always got RC=-1.  

The yield() method calls the cycle() method to check on any work and send PINGREQ,  the return code from keepalive() https://github.com/eclipse/paho.mqtt.embedded-c/blob/develop/MQTTClient/src/MQTTClient.h#L608  is ignored.  That must be fixed.

Furthermore, when error has occurred such as network error, the connect state is not changed.  As a consequence, the return code from isConnected() is still true.

miketran78727, 2017-05-17:
I am testing my patch,  looks like this:

````
template<class Network, class Timer, int a, int b>
int MQTT::Client<Network, Timer, a, b>::yield(unsigned long timeout_ms)
{
    int rc = SUCCESS;
    Timer timer = Timer();
    timer.countdown_ms(timeout_ms);
    while (!timer.expired())
    {
        if (cycle(timer) < 0)
        {
            rc = FAILURE;
	    ping_outstanding = false;
	    isconnected = false;
            break;
        }
    }
    return rc;
}
````
and

````
int MQTT::Client<Network, Timer, MAX_MQTT_PACKET_SIZE, b>::cycle(Timer& timer)
{
....

    keepaliveRC = keepalive();
exit:
    if (rc == SUCCESS)
        rc = packet_type;
    if (packet_type == FAILURE)
        rc = keepaliveRC;
    return rc;
}
````

icraggs, 2017-07-07:
Hi Mike.  I've added some initial tests, and as a result restructured the code that affects this functionality.  I think what I've done so far covers these situations correctly.  If you want to check this out, that would be great, otherwise I'll rely on the tests.

icraggs, 2017-07-14:
Tests and changes made.

closed 2017-07-14 by icraggs

PhilLaFayette, 2017-09-14:
@icraggs & @miketran78727 
What about if the client yields, yet there is no received packet for the client? Won't this force a closeSession(); ?

IE: (pseudo)
`client.connect(connectData);`
`client.subscribe(subTopic, MQTT::QOS0, messageArrived);`
`client.yield(1000);`
`if (client.isConnected())`
`{`
`    // We'll never get here, as yield() will return rc -1 in the case that`
`    // there are no packets waiting...`
`    // switch (packet_type) goes to exit:, which calls closeSession();`
`}`

Thoughts?