eclipse-paho/paho.mqtt.embedded-c #95: Does the Paho MQTT Embedded C supports the TLS feature

PravallikaKG, 2017-06-08:
I fetched the code from https://github.com/eclipse/paho.mqtt.embedded-c''
I am using paho client on my tiny embedded device. The programming is to be done C. Embedded device is of linux flavour. So, i decided to used the library of Paho MQTT Embedded C.

But we i saw the structure of ''MQTTPacket_connectData'', it doesnt have members for certificates or key to support TLS. Does the Paho MQTT Embedded C supports the TLS? Please help me out how to achieve TLS functionality using it?

And also i saw a another library which is in C from "https://github.com/eclipse/paho.mqtt.c". But i think this is for Windows and linux, arm based. I think this i cannot use for my tiny embedded device. Please correct me if my understanding is wrong.

eden-wang, 2017-06-15:
- I don't think it support TLS though the official website declares it does.
- [https://github.com/eclipse/paho.mqtt.c](https://github.com/eclipse/paho.mqtt.c) is for Posix Linux and Windows
-  [https://github.com/eclipse/paho.mqtt.embedded-c](https://github.com/eclipse/paho.mqtt.embedded-c) is for embedded devices

sshtel, 2017-07-07:
When you use library, you can establish TLS tcp connection by yourself and init MQTT stack by passing the socket fd.

icraggs, 2017-07-07:
I wrote an article about using TLS with the embedded client some time ago: 
http://modelbasedtesting.co.uk/2015/01/19/using-tls-with-the-paho-mqtt-embedded-c-client/

We didn't have an actual example in the repository because of licensing - all the TLS implementations were GPL, which is incompatible with the Eclipse Foundation licenses.  Now, however, there is at least one Apache licensed TLS implementation, and I would get an example using that into Paho.

If it's ok, I'll change the title of this issue to "Add TLS implementation example" and use it to track progress.

brimston3, 2017-09-12:
Linux mbedTLS example code for paho.mqtt.embedded-c/MQTTClient (C++ version):
https://gist.github.com/brimston3/aaf728d357ff4d6ac76a871e02adb8c6

shpasser, 2017-09-17:
Thanks a lot @brimston3 it is very helpful.

wuhaogs, 2023-04-08:
@brimston3 ，您好，请问，针对paho.mqtt.embedded-c/MQTTClient-C（C语言版本），有支持 Linux mbedTLS 的示例代码吗？