eclipse-paho/paho.mqtt.embedded-c #99: Q) Shouldn't PING REQ packet be sent if packet is transferred within keepaliveInterval?

sshtel, 2017-07-19:
Thanks for new version release and I appreciate that your work.

I am using paho mqtt embedded-c library for my company's production.
Me and my team are arguing that if the PING REQ should be sent even though there were packet being sent within keep alive Interval time.

For example, keepaliveInterval == 30 and we are sending packet(PUB, QoS==0) at every 10 seconds.

According to library implementation, MQTTClient c->last_sent  value is refreshed when a packet is sent.
In this case, PING packet is never triggered.

Some people are arguing that PING should be sent exclusively even if any packet is sent within keep alive Interval.

Which one is right?

icraggs, 2017-07-19:
The answer is - it depends.

That's why I recently changed the embedded-C library to match the C++ one,  and use two timers, one for tracking sent packets and the other for tracking received packets.  There are two uses for pings:

1) To stop the server closing the session if no other packet has been sent within the keep alive interval.  For this purpose, the last_sent timer is used.

2) To detect a server failure in the event that no other packet is received from the server within the keep alive interval.  For this purpose, the last_received timer is used.

If the application sends QoS 0 publish packets regularly within the keepalive interval then the first condition is satisfied, but not the second.  If QoS 0 publish packets are received as well, then the second condition is satisfied too.

If your application only sends QoS 0 packets but does not receive any then I would expect the ping to be sent.   The condition in keepalive() says:

  if (TimerIsExpired(&c->last_sent) || TimerIsExpired(&c->last_received))

that should send the ping whether the last_sent OR the last_received timer has expired.   If you think this is not happening, let me know.

sshtel, 2017-07-20:
Great, clear now!! Thanks for answer!

closed 2017-07-20 by sshtel