eclipse-paho/paho.mqtt.embedded-c #101: MQTTYield returns -1 in FreeRTOS TCP/IP stack

NarekGh, 2017-07-25:
`FreeRTOS_read` returns 0 in [MQTTFreeRTOS.c](https://github.com/eclipse/paho.mqtt.embedded-c/blob/master/MQTTClient-C/src/FreeRTOS/MQTTFreeRTOS.c), when timeout has accrued, which caused returning of -1 value of `MQTTYield`.  I think it would be right to add a handler for  this  case in `FreeRTOS_read`

```c
	do
	{
		int rc = 0;

		FreeRTOS_setsockopt(n->my_socket, 0, FREERTOS_SO_RCVTIMEO, &xTicksToWait, sizeof(xTicksToWait));
		rc = FreeRTOS_recv(n->my_socket, buffer + recvLen, len - recvLen, 0);
		if (rc > 0)
			recvLen += rc;
		else if (rc < 0) //ERROR case
		{
			recvLen = 0;// WAS rc  
			break;
		}
		else if (rc == 0){ //timeout case
			recvLen = -1;
			break; 
		}
			
	} while (recvLen < len && xTaskCheckForTimeOut(&xTimeOut, &xTicksToWait) == pdFALSE);
```

Has anybody any idea about this issue?

andrew0x0, 2017-08-22:
Network program will return 0 meaning Tcp/Ip layout timeout and MQTTYield seems timer of MQTT layout,it's two different things.

NarekGh, 2017-08-22:
It's really  two different things!!! There is no case to handle timeout in FreeRTOS_read() (FreeRTOS TCP/IP stack).

PhilLaFayette, 2017-09-21:
Please note #110, for if there aren't any messages waiting for the client, then `yield()` will return -1 even if the connection is valid.