eclipse-paho/paho.mqtt.embedded-c #102: MQTT version 5.0 support

icraggs, 2017-07-26:
The MQTT version 5.0 specification is nearing completion, with the availability of working draft 15 (https://www.oasis-open.org/committees/documents.php?wg_abbrev=mqtt&show_descriptions=yes), which is going to be, with very minor changes, the first Committee Specification Draft (CSD).  This means that we are nearing the end of the process of creating the MQTT 5.0 specification.  There is a 30 day review period for the CSD once it's officially published, then if changes are needed as a result of feedback, there will be a second CSD with a further review period, and so on.  We anticipate that two CSDs will be sufficient.

To facilitate MQTT v 5.0 adoption and awareness in the community, and give feedback to the OASIS TC within the CSD review period, James Sutton and I are proposing to start work on implementations in Eclipse Paho.  The rough plan is:

1) Ian to write a Python "test" broker to enable client testing and a server implementation model.

2) James to start work on a Java client implementation.  This will probably be a completely new codebase, because the existing Java client was for a long time deliberately constrained to the Java 1.4.2 language level to be compatible with JavaME.  A new code base will enable later Java language features to be fully exploited.

3) Ian to write C and embedded C client implementations.  I haven't decided yet whether or how to extend the current codebases or write new ones.  My plan is to try extending, and if that proves excessively complex, to review that approach.

4) After Java, James will probably look at JavaScript next.

markusschaber, 2018-07-16:
What's the exact progress on this? 5.0 is officially released since May, and other implementations are catching up...

Thanks!

icraggs, 2018-07-16:
I've added support to MQTTPacket, the higher level libraries will be later this year because I've been concentrating on the mainline Paho C library first.

MQTT 5.0 was not officially finished in May - some commentators have been misleading in that regard.  We have reached Committee Specification 02 (http://docs.oasis-open.org/mqtt/mqtt/v5.0/cs02/mqtt-v5.0-cs02.html), and are now soliciting statements of use.  It is true that there are unlikely to be substantive changes to the specification from now on, but the standardization process is not likely to be complete until 3rd or 4th quarter this year.

fractalembedded, 2021-08-21:
Any update on 5.0 support in this library? Looks like the desktop C/C++ library got support a while ago.

tiaan-lg, 2021-12-09:
What is the update on MQTT v 5.0 Support?

troica65, 2022-02-18:
You mentioned in the initial post that you intended to work on an embedded version of the MQTT client. What happened to that ? Any progress ? What are the biggest obstacles

Singa17, 2024-05-08:
ResponseTopic should work as expected for code in the develop branch. I have added samples that use properties: https://github.com/eclipse/paho.mqtt.embedded-c/blob/develop/MQTTPacket/samples/v5pub0sub1.c#L221

is this project still active?

Yes, it is! I am actively working together with @icraggs on wrapping up MQTTv5 support that was already available for MQTTPacket in the develop branch. I am a developer with Microsoft Azure IoT and my work is sponsored by our group. Next on my list is to create a MQTTClientv5-c equivalent.

[CIPop](https://github.com/CIPop)
Hi CIPop, Any update on the MQTTClientv5-c?