eclipse-paho/paho.mqtt.embedded-c #107: (mbedOS porting) Is MQTTClient thread safe?

maiorfi, 2017-09-11:
I posted issue detail here:

https://developer.mbed.org/questions/79004/Is-MQTTClient-thread-safe/

icraggs, 2017-11-08:
The MQTTClient class is not thread safe in that it is not intended that multiple threads use it simultaneously. 

The MQTT library has however now been updated which might help solve your problem.

I have intended to create a thread safe library, MQTTAsync, but that's not yet done.

closed 2017-11-08 by icraggs

horchi, 2019-08-15:
Hello, is there any thread save c or c++ client impementation? 
I have two different usecases where i need a MQTT client, one is threaded and at one I like to  fork a child. 
I tried the C phao MQTT library but it is not working at all when i open the connection in a process an try opening another connection after fork in the child i get a socke error.

Best regatds
Jörg