eclipse-paho/paho.mqtt.embedded-c #110: MQTT yield() method always returns -1 [Re-visited]

PhilLaFayette, 2017-09-21:
I'm creating a new issue for visibility, as #91 is currently closed.

I'll recreate my [post in that thread](https://github.com/eclipse/paho.mqtt.embedded-c/issues/91#issuecomment-329416943) here:

@icraggs & @miketran78727 
What about if the client yields, yet there is no received/waiting packet for the client? Won't this force a closeSession(); ?

Example: (pseudo)
`client.connect(connectData);`
`client.subscribe(subTopic, MQTT::QOS0, messageArrived);`
`client.yield(1000);`
`if (client.isConnected())`
`{`
`    // We'll never get here, as yield() will return rc -1 in the case that`
`    // there are no packets waiting...`
`    // switch (packet_type) goes to exit:, which calls closeSession();`
`}`

Thoughts?

> Furthermore, when error has occurred such as network error, the connect state is not changed. As a consequence, the return code from `isConnected()` is still true.

Per my case above, wouldn't we want `isConnected()` to return true after a `yield()`, assuming:
* No network error has occurred
* No messages are available to be received

Possible fix: Perhaps add a `packet_type` '`NOMSGS`' case to the `cycle()` switch, or maybe set a flag when `yield()` calls `cycle()`?

brimston3, 2017-09-21:
That's what 0 is supposed to mean in that switch block. ([switch comment re: nomsgs](https://github.com/eclipse/paho.mqtt.embedded-c/blob/develop/MQTTClient/src/MQTTClient.h#L590)). If the timer expires while waiting for a message, your implementation of ipstack.read() must return 0.

charlesoram, 2017-11-09:
The version of IPStack::read() in MQTTClient/samples/linux/main.cpp has this problem (i.e. returns -1 when there are no bytes to be read, causing it to disconnect), whereas the version in MQTTClient/src/linux/linux.cpp works (returns 0 when there are no bytes to read).

rweller59, 2017-11-17:
I'm using embedded library with espidf. I have created a freetros task that calls MQTTYield:
MQTTYield(&mqtt_client, 100);
sleep(1000);

When the task is called, MQTTYield does not return until the callback is triggered with a message arrival. It then returns, the task sleeps and is called again (as scheduled). It then sits at MQTTYield again until the next message arrives.

This is not the behavior I was expecting. Am I mis-understanding something?

rweller59, 2017-11-25:
My issue was not properly defining my timing functions in mqttclient - so this should be closed

rweller59, 2017-11-25:
Could someone provide insight on how/where to modify the IPStack::read() function to return 0 when not bytes are read

alexbondarevhisky, 2018-12-17:
is there an update for this? it's been over a year with the status "open"?
I'm still getting random "disconnected" states after n calls to MQTT yield (the socket is definitely not closed since I'm able to call MQTTConnect again on the same socket).

I'm using the Linux variant of the code.

abondarev84, 2018-12-17:
is there an update for this? it's been over a year with the status "open"?
I'm still getting random "disconnected" states after n calls to MQTT yield (the socket is definitely not closed since I'm able to call MQTTConnect again on the same socket).

I'm using the Linux variant of the code.

icraggs, 2018-12-17:
Are you using the C or C++ version?  The originator of this issue was using the C++ version, by the looks of it.

abondarev84, 2018-12-17:
I'm using the C version, forgot to add that.

abondarev84, 2018-12-27:
from what I was able to understand from the code:

in case yield calls recv, through cycle, (in C case it's the linux_read function) , has no data, the recv fucntion returns -1 and that value is returned as the value from cycle.
the switch(packet_type), -1 goes to the default case which directs to exit tag that marks the session as closed if rc is anything but SUCCESS.

shouldn't the return value in that case be 0 instead of the same as the recv function (which is -1)?

abondarev84, 2018-12-30:
I've changed the return code to reflect the socket returning EAGAIN & EWOUDBLOCK to return 0 from read_socket but I still see the client context chowing as disconnected randomly.

I'll wait for a solution from the developers.

p2k-ko, 2019-11-06:
Is there an update on this Issue?