eclipse-paho/paho.mqtt.embedded-c #115: Timeout-based disconnects sending QOS replies

brimston3, 2017-10-24:
With the linux version of timer and QOS1 or QOS2, it's possible to relatively easily generate a disconnect by receiving a lot of PUBLISH messages with a short yield timeout (like 10ms). It's possible for the timer to be significantly depleted waiting for readPacket() and by the time it wants to send the PUBACK or PUBREC message, the timer itself is expired. This results in no passes through sendPacket's while loop, which in turn returns rc = FAILURE.

Instead of treating it as a timeout, it is treated as a failed packet write, closing the session (cycle() rc = FAILURE; goto exit; ).

This is also possible handling PUBREC and PUBACK messages for QOS2.

Potential workaround: if the timer expires with 0 bytes sent, noop the PUBACK/PUBREC and wait for the server to resend it, but do not fail. If the timer expires during the loop; fail as usual -- odds are you're going to be able to write the very short packet for PUBACK, PUBREC, PUBREL, or PUBCOMP.

icraggs, 2017-11-09:
Are you using MQTTClient or MQTTClient-C?  I've added some timing changes to MQTTClient which  I hope will help this situation.

brimston3, 2017-11-11:
MQTTClient (C++). I've been on the master branch, not develop. Commit 00c5ac9 is what I needed.

closed 2017-11-11 by brimston3