eclipse-paho/paho.mqtt.embedded-c #116: [C++ client] Parameter errors causes connection drop

Novakov, 2017-11-10:
I noticed that MQTTClient treats all error conditions the same and handles them closing session. However few errors can be casued by mistake in parameters in which case session is still working properly. 

For example:
1. Call `client.subscribe("/test", MQTT::QOS0, nullptr);`
1. Try to publish message - fail with not information about error cause (beside "there was an error")

It's pretty hard to spot the cause-effect relationship between these two actions.

It would be nice to get more precise error codes from `MQTTClient`, at least to be able to differentiate between parameters error and network-level errors.

icraggs, 2017-11-11:
This client is meant to be really lightweight, to use as few resources as possible for when it needs to run on a microprocessor with 16k RAM for instance.  Adding more checks takes more resources - that's why they aren't there.  More checks could be added in conditional code (MQTT_DEBUG perhaps) sections so that they can be compiled out if preferred.

Novakov, 2017-11-11:
The checks are there but all lead to the same result: error code and session close. That way you don't know if you messed up with parameters which is easy to fix or there is something wrong with network stack which is much harder to debug.

CIPop, 2023-07-19:
One idea is to:
1. Use preconditions for argument checking. These are equivalent to an assert and can be removed through compilation flags (only used during initial development).
2. Add more logging that can be opt-in (disabled through compilation flags). Logging should be a callback-based system that allows the developer to implement the listener (which might be printf, UART, ARM ITM, etc).
3. Use asserts (which should be disabled in Release builds) to verify intermediate processing steps.

CIPop, 2023-07-19:
Similar request for MQTTClient-C: #111