eclipse-paho/paho.mqtt.embedded-c #117: Mqtt client not receiving wildcard topics

rweller59, 2017-11-17:
If I subscribe to "test", the callback gets called with new messages published to topic 'test'
If I subscribe to "test/+", MQTTYield is released but callback is not triggered for any messages. Message formats tried include 'test/1', 'test/+', 'test/', etc. I realize that only the first one is valid.

This may be related to another issue I thought I had submitted but can't find. That issue is:
MQTTYield is called but does not return until a message is received.
My code is now
  while(1) { MQTTYield(&client, 1000); }
I had this on a timed task but since it never yields, I'm now at this in order to get any messages processed

rweller59, 2017-11-17:
Trying to dig through the code to figure out MQTTYield first. Found comments in MQTTClient.h stating:
/* The Timer structure must be defined in the platform specific header,
 * and have the following functions to operate on it.  */...

Could someone please provide some insight as to what this might look like.

icraggs, 2017-11-18:
Yep - in MQTTClient-C/src/linux/MQTTLinux.c there are examples of those functions.

rweller59, 2017-11-19:
I've gotten a bit further but assuming I still have a setup issue (C++, ESPIDF):
 - MQTTYield does not return until a message arrives
 - wildcards (topic/#, topic/+) are recognized (MQTTYield releases) but no topic/message attached. If I send a message that exactly matches the topic, topic & message are attached

rweller59, 2017-11-21:
Spent a lot of time trying to figure this out but I'm stuck. I believe I've got everything defined properly but received messages are not being sent back to the handler.
 - topic is subscribed 
 - c->messageHandlers[i].topicFilter = topicFilter; is executed in MQTTSetMessageHandler
The first time MQTTYield is called, it calls cycle, it calls readPacket(c, timer); which returns -1 so exit is called which calls MQTTCloseSession which clears the messageHandlers.topicFilters.

As messages are received against the subscribed topic, the cycle function recognizes them and drops into case PUBLISH but since all of the topicFilters are now cleared, the received topic is not processed.

Can you please give me some insight as to what is going on here.

closed 2017-11-21 by rweller59