eclipse-paho/paho.mqtt.embedded-c #120: C++ Message Handler

shpasser, 2017-12-11:
The current implementation of C++ client allows to define C-style message handlers for subscriptions. Is it possible to create those handlers dynamically in runtime?

denravonska, 2018-01-24:
The underlying implementation uses the function pointer class `FP` so if the `subscribe` method accepted an FP object instead it would work as is.

shpasser, 2018-01-28:
Sorry, but I do not see how #109 solves my problem. If I would like to create multiple subscriptions using multiple instances of the same class as their handlers would it work?

brimston3, 2018-01-30:
Yes it would. If you take the example from the PR and make the following changes, it would dispatch with appropriate instances of the same class:

```
...
C1 c1,c2;
decltype(c)::MessageHandlerType fp1,fp2;
fp1.attach(&c1, &C1::mRcv);
fp2.attach(&c2, &C1::mRcv);
...
```

I eventually converted from FP.h to std::function which potentially has some involved special case performance gotchas and a dependency on C++11 (afaik, only my example usage depends on C++11 in #109). I'm not sure it's worth submitting a PR for the std::function version since this does the job. Syntax is nicer though.

icraggs, 2018-03-05:
I'll merge in #109 if the IP checks are resolved.  If that doesn't happen in the next month, then I'll reimplement those changes myself.  It wasn't my intention that this usage should be prevented, but I guess I just hadn't tried it out.

Using std::function might be a good idea, or it might not be.  I tried to limit the dependencies on external libraries, even if standard, because as this library is targeted at constrained environments, it's more likely to encounter older compiler and/or OS versions.

icraggs, 2018-03-17:
The IP checks in #109 didn't pass, so I couldn't merge it.  However, I've added my own implementation in the commit referenced here, in the develop branch.  It includes a test and sample program.  I'll be interested in any comments.

denravonska, 2018-04-05:
@icraggs Your fix worked perfectly for me.

brimston3, 2018-04-09:
I verified this works for me as well. Sorry about the IP thing, the other user added a missed return statement that I should have added myself instead of merging.

closed 2023-11-29 by icraggs