eclipse-paho/paho.mqtt.embedded-c #129: Cannot receive more than 3 messages

PhilLaFayette, 2018-03-28:
Hello. It appears the embedded C client is unable to receive more than three persisted/offline messages.

Please consider the following messageArrived handler:

```
void messageArrived(MQTT::MessageData& md) {
    MQTT::Message &message = md.message;
    Serial.print("Message ");
    Serial.print(++arrivedcount);
    Serial.print(" arrived: qos ");
    Serial.print(message.qos);
    Serial.print(", retained ");
    Serial.print(message.retained);
    Serial.print(", dup ");
    Serial.print(message.dup);
    Serial.print(", packetid ");
    Serial.println(message.id);
    
    Serial.print("Payload ");
    Serial.println((char*)message.payload);
}
```

Consider the following payloads, all of the same topic and QoS=1:
```
payload1 = "foo"
payload2 = "bar"
payload3 = "baz"
payload4 = "qux"
```

While client is offline, sending payloads 1-3, and on client connection, everything works fine.

While client is offline, sending payloads 1-4, and on client connection, program execution stops after `Payload baz`

icraggs, 2018-04-03:
I've added a test (to the develop branch), which queues up a number of messages and receives them.  5 QoS 1 messages to start with.  It passes on Linux (https://travis-ci.org/eclipse/paho.mqtt.embedded-c/builds/361593826).  Perhaps you could see what differences there are between that test and your implementation?

icraggs, 2018-04-13:
I trust that example helped.  You can reopen if not.  Thanks.

closed 2018-04-13 by icraggs