eclipse-paho/paho.mqtt.embedded-c #132: How to compile the program on embedded board

chenchuangchn, 2018-04-08:
Hi~
I've compiled successfully on my linux PC. Now I want to run on my embedded board which is arm architecture. But the resource of my board is not enough  compared with my PC. There is no cmake, no python environment and so on. So how to compile the program on embedded board, Do I specific the cross compile(I can't find) on PC, or other way?
Thank you!

icraggs, 2018-04-13:
It depends on the tools available for that board.  You should be able to find some advice available by internet search.  Cross compiling from Linux is a common approach.  CMake does support cross-compiling too, although I haven't used it myself yet:

https://cmake.org/Wiki/CMake_Cross_Compiling

chenchuangchn, 2018-04-17:
Thanks for your answer. I have knew the code well. And I have found the build.sh in sambles/linux. I think that it will work well to modify gcc to arm-linux-gcc, although I haven't used it myself yet

closed 2018-04-20 by chenchuangchn