eclipse-paho/paho.mqtt.embedded-c #136: MQTTPublish failed to transmit Payload of bigger length

Kvu23, 2018-05-18:
Hello Sir,
                  I am currently using paho mqtt embedded c library in my project. In this i am using MQTT 3.1.1 embedded Client. i am tramsmitting JSON payload here to mosquitto broker here. But I am Unable to transmit JSON payload of the Bigger length lets say some 1 KB....MQTTPublish is failing with the rc = -1. in my case. if i am crossing payload length greater than 67 bytes MQTTPublish is failing in my code.
How i can resolve this....how i can send Bigger payload in this case......Please Do help me....I m stuck and i dont have any clue.....

FlandreUNX, 2018-05-18:
Try to increase the RAM space for buffer

gparmeggiani, 2018-05-21:
Hello, when you create an instance of the MQTT client, you can pass a custom size for the MQTT packet using the thirt parameter of the template:
`template<class Network, class Timer, int MAX_MQTT_PACKET_SIZE = 100, int MAX_MESSAGE_HANDLERS = 5>
class Client
{
//...
`

As you can see the default is set to 100 bytes.

You can pick a different size like so:
`MQTT::Client<IPStack, Countdown, (5*1024)> client; //5Kb
`

Hope that helps

Kvu23, 2018-05-22:
i am using paho mqtt embedded c library code in that where i can increase
MAX_MQTt_PACKET_SIZE?   can you help on this regarding this....?

FlandreUNX, 2018-05-22:
@Kvu23 
See the code below, this is the C

`MQTTClientInit(&mqtt, &mqtt_Network, 30000, mqtt_SendBuffer, TX_BUFFER_MAX, mqtt_ReadBuffer, RX_BUFFER_MAX);`

The mqtt_SendBuffer[TX_BUFFER_MAX] & mqtt_ReadBuffer[RX_BUFFER_MAX],
The buffer size is defined by yourself and passed into the initialization function

Kvu23, 2018-05-22:
Thanks... a lot.....i think this will work......i will check it out and
update you.....

icraggs, 2018-05-30:
@FlandreUNX @gparmeggiani thanks for helping out!    

@Kvu23 can I assume this is working for you now?

Kvu23, 2018-05-30:
yes this is fixed

Kvu23, 2018-05-30:
working fine you resolved my issues.....
thanks a lot

closed 2018-05-30 by icraggs