eclipse-paho/paho.mqtt.embedded-c #138: MQTTYield Doesn't send an PUBACK back to the broker.

hiteshpratyushv, 2018-05-29:
I am running a MQTT client of ESP32 and it receives the published message but it doesn't reply with a PUBACK so the broker sends the message again.
I am using MQTTYield to get message which is published to it

icraggs, 2018-05-30:
Do you know what the size of the message is that is being sent to you?  Have you initialized the receive buffer in MQTTClientInit to be big enough?

hiteshpratyushv, 2018-05-30:
The message size that is being sent is 5 bytes and the buf size is 512 bytes.
You can get the whole code in the github link below.
https://github.com/hiteshpratyushv/ESP32_MQTT_GAP_GATEWAY

icraggs, 2018-05-30:
I would suggest putting debug statements in the cycle function, and/or the TCP read, to see if the publish packet is received correctly first.

hiteshpratyushv, 2018-05-30:
It is being recieved as the message handler prints it out in the log.
But idk why the ACK is not being sent back.

icraggs, 2018-05-30:
In that case, in MQTTClient.c, following deliverMessage()  there is the following code:

if (msg.qos != QOS0)
            {
                if (msg.qos == QOS1)
                    len = MQTTSerialize_ack(c->buf, c->buf_size, PUBACK, 0, msg.id);
                else if (msg.qos == QOS2)
                    len = MQTTSerialize_ack(c->buf, c->buf_size, PUBREC, 0, msg.id);
                if (len <= 0)
                    rc = FAILURE;
                else
                    rc = sendPacket(c, len, timer);
                if (rc == FAILURE)
                    goto exit; // there was a problem
            }
            break;

which sends back the ack, in the case of QoS 1 or 2.  You should check the return value from sendPacket, which calls the underlying OS-specific TCP routines.

hiteshpratyushv, 2018-05-30:
What if I just change QOS to 0 ?
Does it matter ?
Also the mosquitto_pub client I am using to publish messages is on QOS 0

closed 2018-05-31 by hiteshpratyushv

icraggs, 2018-05-31:
FYI - there are no acks with QoS 0.

hiteshpratyushv, 2018-05-31:
No problem.
I have figured it out and I have no problem using QOS 0