eclipse-paho/paho.mqtt.embedded-c #139: mqtt packet id

msnistal, 2018-05-30:
Hello, could someone tell me if the packet id is unique for each client or for each type of packet control? That is, if I am a client, I have to have a package counter for each type of message (pub, sub, unsub) or a single counter per client.

Thank you very much

FlandreUNX, 2018-05-30:
You are right that there is a variable in the client structure as an ID counter.
In "typedef struct MQTTClient"
See "unsigned int next_packetid"

msnistal, 2018-05-30:
Thank you! So there is just a counter for ALL the messages sent with Id
rigth?

*MªSaleta Nistal*
Electronic Design Engineer

Exploded View, SL
(+34) 665328715
msnistal@explodedview.es

Confidentiality notice:
This e-mail message, including any attachments, may contain legally
privileged and/or confidential information. If you are not the intended
recipient(s), or the employee or agent responsible for delivery of this
message to the intended recipient(s), you are hereby notified that any
dissemination, distribution, or copying of this e-mail message is strictly
prohibited. If you have received this message in error, please immediately
notify the sender and delete this e-mail message from your computer.

Este mensaje de correo electrónico y sus documentos adjuntos están
dirigidos exclusivamente a los destinatarios especificados. Si usted recibe
este mensaje por error, por favor comuníqueselo inmediatamente al remitente
y elimínelo ya que usted no está autorizado al uso, revelación,
distribución, impresión o copia de toda o alguna parte de la información
contenida. Gracias.
De conformidad con la Ley Orgánica de Protección de Datos de Carácter
Personal 15/1999, le recordamos que sus datos han sido incorporados en un
fichero de datos de carácter personal del que es titular EXPLODED VIEW, S.
L. U, debidamente registrado ante la AEPD y cuya finalidad es la gestión de
las comunicaciones internas y externas vía telemática. Así mismo, le
informamos que sus datos no serán cedidos a terceros sin su consentimiento.
Puede ejercitar sus derechos de Acceso, Rectificación, Cancelación y
Oposición en C/ MARIA IGNACIA, 8 - 28039, MADRID (Madrid) o enviando un
correo electrónico a laia@explodedview.es.

msnistal, 2018-05-30:
> The packet id is unique for each client, not for each packet.
>
> Also note, that the broker and client each use separate packet ids: the
> sender in each session uses the packet id to keep track of responses.

Thank you very much in deed. First time I use this way of communicating and
its really helpful. Thank you very much!

*MªSaleta Nistal*
Electronic Design Engineer

Exploded View, SL
(+34) 665328715
msnistal@explodedview.es

Confidentiality notice:
This e-mail message, including any attachments, may contain legally
privileged and/or confidential information. If you are not the intended
recipient(s), or the employee or agent responsible for delivery of this
message to the intended recipient(s), you are hereby notified that any
dissemination, distribution, or copying of this e-mail message is strictly
prohibited. If you have received this message in error, please immediately
notify the sender and delete this e-mail message from your computer.

Este mensaje de correo electrónico y sus documentos adjuntos están
dirigidos exclusivamente a los destinatarios especificados. Si usted recibe
este mensaje por error, por favor comuníqueselo inmediatamente al remitente
y elimínelo ya que usted no está autorizado al uso, revelación,
distribución, impresión o copia de toda o alguna parte de la información
contenida. Gracias.
De conformidad con la Ley Orgánica de Protección de Datos de Carácter
Personal 15/1999, le recordamos que sus datos han sido incorporados en un
fichero de datos de carácter personal del que es titular EXPLODED VIEW, S.
L. U, debidamente registrado ante la AEPD y cuya finalidad es la gestión de
las comunicaciones internas y externas vía telemática. Así mismo, le
informamos que sus datos no serán cedidos a terceros sin su consentimiento.
Puede ejercitar sus derechos de Acceso, Rectificación, Cancelación y
Oposición en C/ MARIA IGNACIA, 8 - 28039, MADRID (Madrid) o enviando un
correo electrónico a laia@explodedview.es.

msnistal, 2018-05-31:
Thank you!
 One more thing about packet id:
in the Oasis MQTT version 3.3.1 publication stands:
"After it has sent a PUBACK Packet the Receiver MUST treat any incoming PUBLISH packet that contains the same Packet Identifier as being a new publication, irrespective of the setting of its DUP flag. "
Is that correct? What if the PUBACK message never gets to the Pub sender and he resends the packet with the DUP flag=1? Should it be treated as a new packet and not the old one?
thank you

icraggs, 2018-05-31:
It should be treated as a new one.  That's why QoS 1 messages could be duplicated - QoS 1 == "at least once".  To ensure no duplication, QoS 2 is needed.

closed 2018-11-21 by icraggs

goog, 2019-07-22:
if set qos = 1, and fix packet id to 0, what will happen