eclipse-paho/paho.mqtt.embedded-c #141: MQTT_TASK option cannot work

beamer145, 2018-06-07:
I tried to integrate the stack with the MQTT_TASK option for a FreeRTOS based project but the code is is just wrong for this option. 
The biggest problem is that in MQTTRun there is a while loop that will just lock the mutex forever:
	while (1)
	{    
#if defined(MQTT_TASK)
		MutexLock(&c->mutex);
#endif
		TimerCountdownMS(&timer, 500); /* Don't wait too long if no traffic is incoming */
		cycle(c, &timer);
#if defined(MQTT_TASK)
		MutexUnlock(&c->mutex);
#endif
	}
I tried using the echo sample with this and it never gets past the connect because the mutex is locked.
Not sure how to fix this because it is not really clear what exactly you are trying to do.

Also in the echo sample there is a while loop which will not do the 1s delay in case mqtt task is defined.
	while (++count)
	{
            ...
#if !defined(MQTT_TASK)
           if ((rc = MQTTYield(&client, 1000)) != 0)
           {
              LOGGER_LOG_ERROR( LOG_CLASS,"Return code from yield is %d\n", rc);
           }
#endif
	}

I assume this can be easily fixed by having a delay in an extra #else statement.

zyzyzyzzyy, 2018-08-01:
add  "portYIELD()" or “vTaskDelay()” in the end of MQTTRun whiie loop.

like this
`void MQTTRun ( void* parm ) {
  Timer timer;
  MQTTClient* c = ( MQTTClient* ) parm;
  TimerInit ( &timer );

  while ( 1 ) {
#if defined(MQTT_TASK)
    MutexLock ( &c->mutex );
#endif
    TimerCountdownMS ( &timer, 500 ); /* Don't wait too long if no traffic is incoming */
    cycle ( c, &timer );
#if defined(MQTT_TASK)
    MutexUnlock ( &c->mutex );
#endif
		portYIELD();
  }
}`

SharathKaranth, 2018-12-05:
But after some time(1.5 times keep alive) when pingreq is sent it gets out of the loop and the subscribe fails . Any workarounds for this?