eclipse-paho/paho.mqtt.embedded-c #146: connackData generates faulty return codes on some systems

joncard1, 2018-07-29:
In MQTTClient.h line 755, a pointer to connackData#rc is passed in to MQTTDeserialize_connack as a pointer to a char. connackData#rc is declared an int. This causes the return code to be written to the lowest address byte in the int. This causes the return codes to be represented as 83886080, 67108864, 50331648, 33554432, and 16777216. Presumably, my system has a different endianess than the developer's.

icraggs, 2018-11-02:
More or less an accident.  The MQTTPacket library was written first, in C.   The first MQTTClient layer was written later in C++ and I used a different data type because that seemed better at the time.  I wasn't thinking of the conversion, and where I tested it it worked.  

As we have semantic versioning on the projects, changing the external interface to MQTTPacket means bumping the most significant version number, which I didn't want to do just yet.

So please do submit a PR.  Thanks.  And sorry for the delay in replying - I've had some other projects to take care of too.

joncard1, 2018-11-04:
I'll put in a PR soon. Thanks.