eclipse-paho/paho.mqtt.embedded-c #149: Unsubscribing topic returns fail

rahulbpatel8, 2018-08-09:
I have ported this library with esp-idf and using it with linux api platform. The issue I am facing is that when using MQTTUnsubscribe() from separate running task, it returns most of the time failure. but while using MQTTUnsubscribe() in subscribed message handler, it works fine.
After some debugging in the library, I see that UNSUBACK is in the state machine runs in cycle but checks in waitfor() function gets failed in MQTTUnsubscribe(). 
  
    if (waitfor(c, UNSUBACK, &timer) == UNSUBACK)
    {
        unsigned short mypacketid;  // should be the same as the packetid above
        if (MQTTDeserialize_unsuback(&mypacketid, c->readbuf, c->readbuf_size) == 1)
            rc = 0; 
    }
    else
        rc = FAILURE;

while UNSUBACK cycle() already returns UNSUBACK to waitfor().

This causes MQTTUnsubscribe() fails. but this issue occurs only if the its called from other task then registered message handler. Is it related to context of two different task. I tried with locking use of cycle(), but same result.

Ritesh236, 2018-08-10:
Hi Developers,

If anyone has faced this type of issue before then please update to us to resolve this issue ASAP.

icraggs, 2018-08-10:
I presume that waitfor() returned with an error code?  What is it?

Are you using the client mutex to block more than one call running simultaneously?

rahulbpatel8, 2018-08-10:
waitfor() returned with 0. and mutex is used with MQTTYield() and waitfor() where cycle() is used for mutual exclusion of cycle().

Ritesh236, 2018-09-04:
Hi,

Anyone has resolved this issue from MQTT Paho Library to get Unsubscribe event from other than MQTT Task?

Let me know if any fix for this ASAP.

Ritesh236, 2018-09-11:
Hi,

Anyone has any update regarding this issue?

Ritesh236, 2018-09-14:
Hi,

Anyone has any update regarding this issue or any clue for that?

dipenparmar, 2018-09-17:
@icraggs any suggestions for above?

Ritesh236, 2018-09-19:
@icraggs do you have any update regarding this or any suggestions for this issue?

icraggs, 2018-09-19:
So you have

1) a background task started with MQTTStartTask
2) the UNSUBSCRIBE return value from cycle() is consumed in the MQTTRun() function
3) so the MQTTUnsubscribe() function never sees the UNSUBSCRIBE value

is that correct?

Ritesh236, 2018-09-19:
@icraggs, As per last answer provided by Rahul, we are unsubscribing MQTT from other task instead of same MQTT Task.

So, What you suggest to resolve that iseue as we are critical stage for this right now into one of our product?

Ritesh236, 2018-10-04:
@icraggs, is there any suggestion regarding this issue from your side? Do you need any details or any more information regarding this?