eclipse-paho/paho.mqtt.embedded-c #150: C client do not print the data when the buf not enough for the coming data

octGitHub, 2018-08-29:
Hi there, 

    Firstly, thx for the project.
    I was testing the client in c .  I ran the demo with 
     (  paho.mqtt.embedded-c/MQTTClient-C/samples/linux/stdoutsub.c )
    and found "C client do not print the data when the buf not enough for the coming data"
    I suggest the recieve buf in  sample code should be larger 
    or just fix this situation once for all.

*Thx , anyway.*

icraggs, 2018-11-02:
As there is no dynamic memory allocation in the embedded client library, and I've followed that principle in the samples, so the maximum packet size has to be limited by the read buffer.

icraggs, 2018-11-21:
The sample could have a maximum read buffer length parameter.

jack4k, 2018-12-13:
Is there any way to increase package size in paho for embedded-c. using it from esp8266-rtos v3.0, but it look's like it's limited to 128 byte max message size, anything longer, at best it will not write message out or at worst mqtt lose connection.
read and write buffers are set at 2048.

CIPop, 2023-07-19:
Buffer allocation and management are the responsibility of the application:
```c
void MQTTClientInit(MQTTClient* c, Network* network, unsigned int command_timeout_ms,
		unsigned char* sendbuf, size_t sendbuf_size, unsigned char* readbuf**, size_t readbuf_size```

Please check the network layer implementation for any issues that might result in truncating the messages.

icraggs, 2023-11-29:
Answered.

closed 2023-11-29 by icraggs