eclipse-paho/paho.mqtt.embedded-c #151: Linux TimerInit wrong initialization

Tassenpudding, 2018-09-05:
I am not sure because I used the Linux code to adapt to another platform, but the TimerInit shouldn't set the end_time to zero, because if it comes to the comparing Timer Functions there you get "gettimeofday" what is returning the current unix time. The unixtime can't be zero and if comparing with zero always gives the same result.
I post the change i made here, maybe it could be done simpler.
MQTTClient-C\src\linux\MQTTLinux.c
void TimerInit(Timer* timer)
{
struct timeval now;
gettimeofday(&now, NULL);
&timer->end_time.tv_sec = now.sec;
&timer->end_time.tv_usec = now.usec;
}

scaprile, 2018-09-05:
Without actually looking at the client code but just observing the header file and the timer code in the c file, I see the TimerInit() function is just to set the structure to a known value. The timer setting functions are TimerCountdown() and TimerCountdownMS(); both read the current time and add the desired value.
So, my guess is that in your modification you are calling TimerIsExpired() before calling one of the setting functions.
Just setting to 0 is useful to catch errors like yours, so I wouldn't init it to current time.

icraggs, 2018-11-28:
@scaprile is correct.

closed 2018-11-28 by icraggs