eclipse-paho/paho.mqtt.embedded-c #155: Any max dimension on user and password?

Gior80, 2018-10-29:
Hi , 

I'm using lwmqtt for connecting over TLS with JWT auth. 

even if I know for sure that jwt is good ( I tested it above other client ) I'm getting always bad -10 4 error. 
as jwt (used in password) and deviceid are so so long string I would like ask you if there's any limitations on the length

icraggs, 2018-10-29:
Ok, I just looked up what JWT is, and I assume it's JSON web token.

Depending on which client library you are using, there is a maximum packet size parameter which could be 100 for instance.  Username and password are part of the connect packet, so if they are long, they could exceed the max packet size set.  In that case, increase the max packet size.

Gior80, 2018-11-02:
Thanks!

I solved!

icraggs, 2018-11-02:
Great!

closed 2018-11-02 by icraggs