eclipse-paho/paho.mqtt.embedded-c #159: Differences function "sendpacket" in C++ MQTTClient and MQTTClient-C

johannes-f, 2018-12-04:
Hi,
I just want to ask, why there are differences in the "sendpacket" function in 
master/MQTTClient-C/src/MQTTClient.c
and
master/MQTTClient/src/MQTTClient.h
especially in the while loop and timer expired handling.
1. MQTTClient.c: 
`while (sent < length && !TimerIsExpired(timer))
    {
        rc = c->ipstack->mqttwrite(c->ipstack, &c->buf[sent], length, TimerLeftMS(timer));
        if (rc < 0)  // there was an error writing the data
            break;
        sent += rc;
    }
`
2.  MQTTClient.h: 
`
while (sent < length)
    {
        rc = ipstack.write(&sendbuf[sent], length - sent, timer.left_ms());
        if (rc < 0)  // there was an error writing the data
            break;
        sent += rc;
        if (timer.expired()) // only check expiry after at least one attempt to write
            break;
    }
`
Is there a reason why these lines of code differ? There is a good comment behind the timer expired check, why?

We are using MQTTClient-C (from Keil CMSIS (v1.0.0)) and ran into problems, that sendpacket returns -1, because the timer is expired. But "length is 4 and sent is 0". 
We tried to change the yield timeout, but no configuration helps. The only change that fixed all of -1 problems was, changing sendpacket to:
` 
while (sent < length)
    {
        rc = c->ipstack->mqttwrite(c->ipstack, &c->buf[sent], length, TimerLeftMS(timer));
        if (rc < 0)  // there was an error writing the data
            break;
        sent += rc;
        if (TimerIsExpired(timer))
            break;
    } 
`
like in MQTTClient.h

In case I missed any change requests to MQTTClient.c, sorry for that.
Thanks for any information.

Regards,
Johannes

hainguyen8y, 2019-01-02:
Johannes, I have same your problem sometimes. I think it should have a attempt to write before checking the timer expired like the comment in MQTTClient.h. I just convert to the "do while". Everything works well. It's due to the timing race.

d1c0, 2019-03-05:
Hi,
Just confirming the timing issue in sendPacket(). Running with a 10ms yield time brings out the worst in it. With such short latency, there is also a corresponding timing issue in readPacket(). By the time the first one or two bytes are read, the timer has expired and the rest of the packet is lost. hainguyen8y's fix works well for sendPacket(). For receivePacket(), one could ignore TimerLeftMS() and use custom timing. This disrespects the yield timeout, but is stable, and an acceptable workaround when using a pre-emptive RTOS.
Regards,
Richard

edwin-oetelaar, 2019-06-15:
Also notice the other bug? 

c->ipstack->mqttwrite(c->ipstack, &c->buf[sent], **length**, TimerLeftMS(timer)); 
/* Wrong length, should be **(length-sent)** not length */

versus

ipstack.write(&sendbuf[sent], **length - sent**, timer.left_ms());