eclipse-paho/paho.mqtt.embedded-c #166: Is it work without Freertos？

youfu88, 2019-01-16:
Dear:
  Is it work without Freertos？
  I really need your help.
Thank you very much!

scaprile, 2019-01-16:
I assume you want to  know if the library and/or the client runs without an RTOS, on bare metal.
I can answer for the library: yes, it is a set of functions generating and parsing MQTT messages without using "system" functions, with no dynamic memory allocation nor threading, and non-blocking equivalents when you have to wait. I've run it on bare metal on a Cortex-M3 (see the examples).
As for the client... if you don't need a real full-fledged client, you can do just with the packet library as I did. If you do, well, wait for the next one to jump in, I haven't used it (yet?).
The follow up question is what transport mechanism will you be using. I've run it over sockets in Linux (testing) and for real over bare metal UART (connected to a 3G module). You'll have to write the transport for your particular choice here.

CIPop, 2023-07-19:
There are 3 different options when using the library:
- MQTTPacket : packet level encoders only. Application is responsible with implementing and driving the MQTT protocol state machine
- MQTTClient-C: blocking I/O C language client
- MQTTClient: C++ blocking I/O language client

In all cases, your application is responsible with providing the network layer to the broker.
If the broker requires TLS, your application is required to provide a TLS client (one is not provided in the library). There are many TLS client options, depending on your environment, that run either on the MCU (e.g. mbedTLS, WolfSSL, etc.) or on the modem.

icraggs, 2023-11-29:
Answered.

closed 2023-11-29 by icraggs