eclipse-paho/paho.mqtt.embedded-c #172: Issue on MQTTSubscribe() on CC3200 port

micronpn, 2019-03-19:
Hello, I am experiencing problems trying to subscribe a data, everything else is working. 
Investigating the situation I found that it fails in cc3200_write() 
and more precisely following the called functions it fails in _SlDrvDataWriteOp() (returning 0xffff instead of 0) inside sl_Send() (file socket.c) that is quite strange considering that it is called by other functions to connect the broker and publish data. The source of the error is g_pCB->SocketTxFailure that is set to a value != 0. Also this is quite strange but seems it is common also in other situations (I mean for connect and publish) without generating errors.

This is the relevant part of my code:

static unsigned char sendbuf[80], readbuf[80];
static const char topicSub1[] = "home/garden/fountain";
#define SUBSCRIBE_ENABLED  1
static Options_r *s_pR;

void messageHndl(MessageData* md)
{
   MQTTMessage* message = md->message;

   if(s_pR->showtopics)
      Report("%.*s\t", md->topicName->lenstring.len, md->topicName->lenstring.data);
   if(s_pR->nodelimiter)
      Report("%.*s", (int)message->payloadlen, (char*)message->payload);
   else
      Report("%.*s%s", (int)message->payloadlen, (char*)message->payload, s_pR->delimiter);
}
//------------------------------------------------------------------------------

static int32_t MqttClientStart(MQTTClient *pClient, Options_r *pOpts, bool secure)
{
   NetworkInit(&pOpts->network);
   // [MM]to test open connection without security and without authentication
   int32_t retVal = 0;
   if(secure == false)
   {
      if((retVal = NetworkConnectUrl(&pOpts->network, pOpts->host, pOpts->port)) != 0)
         printf("Return code from network connect is %d\n", retVal);
   }
   else
   {  // [MM] to test secure connections using certificates
      SlSockSecureFiles_t *psockSecureFiles = getSelectedBrokerSecureFiles();
      if((retVal = TLSNetworkConnectUrl(&pOpts->network, pOpts->host, pOpts->port, psockSecureFiles, SL_SO_SEC_METHOD_SSLv3_TLSV1_2, SL_SEC_MASK_SECURE_DEFAULT, 0)) != 0)
         printf("Return code from network connect is %d\n", retVal);
   }
   MQTTPacket_connectData data = {
      .willFlag = 0,
      .MQTTVersion = 3,   // [MM] 3 for 3.1, 4 for 3.1.1
      .clientID.cstring = pOpts->clientid,
      .username.cstring = pOpts->username,
      .password.cstring = pOpts->password,
      .keepAliveInterval = 10,
      .cleansession = 1
   };

   MQTTClientInit(pClient, &pOpts->network, 30000, sendbuf, sizeof(sendbuf), readbuf, sizeof(readbuf));
   UART_PRINT("Connecting to %s, port: %d\n", pOpts->host, pOpts->port);
   retVal = MQTTConnect(pClient, &data);
   UART_PRINT("Connected %d\n", retVal);
   return retVal;
}
//------------------------------------------------------------------------------

static void MqttClient(MQTTClient *pClient, Options_r *pOpts)
{

   // Initialze MQTT client
   MqttClientStart(pClient, pOpts, false);

   MQTTMessage msg = {
      .qos = QOS1,
      .retained = 0,
      .dup = 0,
      .id = 1,
      .payload = "Push button sw2 is pressed on CC32XX device",
      .payloadlen = 43
   };

   while(true)
   {
      int32_t retVal = -1;
      bool connected = MQTTIsConnected(pClient);
      UART_PRINT("It is %s\n", (connected == true) ? "connected" : "not connected");
      if(false == connected)
         MqttClientStart(pClient, pOpts, false);

#if  SUBSCRIBE_ENABLED == 1
      if( (retVal = MQTTSubscribe(pClient, topicSub1, pOpts->qos, messageHndl) != 0) )
      {
         UART_PRINT("Return code from MQTT subscribe is %d\n", retVal);
         connected = MQTTIsConnected(pClient);
      }
#endif
      while(true)
      {
         if( false == MQTTIsConnected(pClient) )
         {
            UART_PRINT("Return code from MQTT IsConnected is %d\n", retVal);
            MqttClientStart(pClient, pOpts, false);
         }
         if( (retVal = MQTTPublish(pClient, "topic2", &msg) != 0 ) )
         {
            UART_PRINT("Return code from MQTT publish is %d\n", retVal);
            break;
         }
         if( (retVal = MQTTYield(pClient, 2000)) != 0 )
         {
            UART_PRINT("Return code from yield is %d\n", retVal);
            break;
         }
      }

      UART_PRINT("Stopping\n");
      if( (retVal = MQTTDisconnect(pClient) != 0) )
         UART_PRINT("Return code from MQTT disconnect is %d\n", retVal);
      pOpts->network.disconnect(&pOpts->network);
   }
   UART_PRINT("\r\n Exiting MQTTclient\r\n");
}
//------------------------------------------------------------------------------

icraggs, 2019-03-20:
Hi.  If the other calls are working, the first thing I would make sure of is that the send buffer is big enough to hold the packet data.  It looks like it should be, but just to be sure.

micronpn, 2019-03-20:
No. it is big enough. But I found something probably wrong also if I am not sure to understand what was your intention.
I f you check at MQTTDeserialize_suback(...) you will find it is called only by:
```
int MQTTSubscribeWithResults(MQTTClient* c, const char* topicFilter, enum QoS qos,
       messageHandler messageHandler, MQTTSubackData* data)

```
this is the call:
`if (MQTTDeserialize_suback(&mypacketid, 1, &count, (int*)&data->grantedQoS, c->readbuf, c->readbuf_size) == 1)`

and if you look at its prototype:
`int MQTTDeserialize_suback(unsigned short* packetid, int maxcount, int* count, int grantedQoSs[], unsigned char* buf, int buflen)
`
you see that grantedQoSs[] is a pointer to an array of integers and inside is used alaso as an array:
		`grantedQoSs[(*count)++] = readChar(&curdata);`
without any validiy check on the array size and considering that the only call pass to it a single bucket array I don't figure what was your original intention.
From my point of view MQTTDeserialize_suback() has to be corrected/rewritten because there is a series of potential bugs and bad practices.