eclipse-paho/paho.mqtt.embedded-c #173: cannot receive message when subscribed on dynamic topic

AnumSheraz, 2019-03-21:
I have a String topic
```
String baseTopic  = "arn:aws:iot:us-east-2:xxxxxxxxxxxx:thing";
String espInTopicStr  = baseTopic + "/" + config.machineName + "/espIn"; //config.machineName is a string "myDeviceAbc"
```
I also have the same topic hardcoded as const char*
```
const char* hardCodedTopic = "arn:aws:iot:us-east-2:xxxxxxxxxxxx:thing/myDeviceAbc/espIn";
```
then I subscribe on `espInTopicStr .c_str()` topic
```
   int rc = 99;
   while (rc != 0){
      rc = client.subscribe(espInTopicStr.c_str(), MQTT::QOS0, messageArrived);
      if (rc != 0) {
        Serial.print("rc from MQTT subscribe is ");
        Serial.println(rc);
        return;
      } else {
        Serial.println("");
        Serial.println("MQTT subscribed");
      }
      Serial.println("rc: " + String(rc));
      delay(500);     
   }
```
It gets subscribe successfully. but I don't receive any message on this topic when I publish from AWS console. 
Whereas if I subscribe on hardCodedTopic (which is exactly the same), I do receive published messages. I though maybe due to conversion from String to char array, there might be some unknown characters appended in the char array espInTopic, but I also checked the size and content of both topics which is exactly the same. 
full details of the question
https://arduino.stackexchange.com/questions/62743/mqtt-not-subscribe-on-topic-thats-converted-from-string-to-char-array 

Just want to check if I am doing something wrong, or is there something added in topic during conversion or during passing it to subscribe method ?

Youldboukhitine, 2020-08-06:
Hello @AnumSheraz , 

I had similar issue and fix it. 
There is a mechanism of callback notification on the message received. 
Basically, when a message arrives from a topic a loop checks and compares the topic of the message arriving and the current subscribe topics.
The subscribed topics are not stored it's only a pointer storing the address of your string. 

So when you subscribe to a topic with a string defined locally in a function there a lot of change that you never get a notification because of the local variable declaration/definition location (probably on the stack).

As a workaround try to define in a global variable the Topic name and see the behavior. 

Hope this helps. 

Yacine

icraggs, 2023-11-29:
Answered above.

closed 2023-11-29 by icraggs