eclipse-paho/paho.mqtt.embedded-c #180: Result enum is too common

MikhailNatalenko, 2019-06-28:
Hello!  
I am trying to compile MQTTClient with stm32 HAL, and I have problems with enum redeclaration

```enum returnCode { BUFFER_OVERFLOW = -2, FAILURE = -1, SUCCESS = 0 };```

Isn't `SUCCESS` word too common? There is a big chance to redecorate it. I understand, that it's almost impossible to fix it without loss of support. How can I workaround this problem? I don't want to make fixes in stable libraries.

CIPop, 2023-07-19:
Fixing this would indeed be a breaking API change for existing clients.
I will be considering these issues for MQTTv5 clients.

CIPop, 2023-07-26:
I looked closer and realized that `returnCode` is not actually a required part of the API. 
These values are returned by the code but the return types of the functions are not actually `returnCode`.  Given that apps used the values in the past, one could argue that modifying them is still a breaking change. 

I still recommend we take the change but I would separate it from the V5 work.

icraggs, 2023-07-27:
In the main C client, I prefixed all constants with either MQTTCLIENT_ or MQTTASYNC_, and the V5 reason codes with MQTTREASONCODE_

I'm not saying those are the best choices, but we could use the same convention for the sake of making the APIs similar.

The other thought I had was to prefix them all with PAHO_ to distinguish between other APIs. But I would advocate following the main C client API convention, unless there's a specific objection, to get more of a family feel.

icraggs, 2023-07-27:
If we're going for a significant renaming exercise, which I think we should for these enums/constants, then we'll need a major version change to signify that (2.0.0).