eclipse-paho/paho.mqtt.embedded-c #183: Question. TLS arduino?

dmet6789, 2019-07-07:
Hello
It is possible to use this library to connect with TLS from arduino yun to a MQTT broker??
if it is possible, do you have a little example or a tutorial or some instructions to make a simple example to publish a message from arduino yun using tls?
thanks in advance for your help
best regards
diego

CIPop, 2023-07-19:
The library helps by providing an `MQTTPacket` data transformation layer between strongly typed C structures and what needs to be sent over the wire.
Using this layer, it would be possible, but you would need to implement all the broker logic. Constraining what your broker supports may be achievable using Arduino.

icraggs, 2023-11-29:
Thanks @CIPop

closed 2023-11-29 by icraggs