eclipse-paho/paho.mqtt.embedded-c #196: Question about this repository

sigusr1, 2019-12-07:
Is this repository discarded?

I found repository  [https://github.com/eclipse/paho.mqtt.c](https://github.com/eclipse/paho.mqtt.c) update frequently， but this repository hasn't updated for years.

scaprile, 2019-12-07:
Embedded code can be mature without turning obsolete or considered discarded.
Code here supports MQTT 3.1.1 and it did not change...
However, since MQTT 5 hit the streets this year, we might start to see some traffic.

sigusr1, 2019-12-08:
But it dosn't support Qos 2 publish.
And I think there is lots of work to be done to fully comform to MQTT 3.1.1 .

scaprile, 2019-12-08:
If any or some of the sub-projects described in the README,md file do not satisfy your needs or wishes, you are welcome to participate in the forums mentioned there.
Even better, please pinpoint those issues in specific pieces of code and/or contribute your own code to fix them.

closed 2023-11-29 by icraggs