eclipse-paho/paho.mqtt.embedded-c #203: Not receiving PINGRESP

tjeneh, 2020-01-28:
Hello,

Ive been attempting to connect to AWS Mqtt broker using this library. I have successfully connected and i am communicating until the keepalive expires. after examining the issue I discovered I am either not receiving any PINGRESP from the AWS broker or when the AWS broker responds the client isnt reading it. Has anyone faced this issue or know how I can resolve it? 

Any assistance would be greatly appreciated.
Thank you

scaprile, 2020-01-28:
I would start by solving the 'either' from your sentence.
Have you googled for this ? https://aws.amazon.com/iot-core/pricing/additional-details/
"_**Your device can ping as often as every 30s, with no charge for MQTT PINGREQ or PINGRESP messages. Pinging more frequently may result in the Device Gateway not responding to PINGREQ**_"
Have you tried a desktop client using the very same timing ?
You can sniff the traffic with Wireshark.
You can enable MQTT_DEBUG and trace the code.
Are you sure the PINGREQ is going out ?

tjeneh, 2020-01-28:
thanks for the fast response.

Ive read through the AWS MQTT documentation several times. my keepalive is set to 1200 seconds and I'm not seeing a response to the first ping (I've tried using keep alive values ranging from their min 30 seconds to their max 1200 seconds). 

Ive connected to the AWS broker from a MQTT desktop application and didn't get disconnected.

Ive tried using Wireshark and expected to see MQTT or TLS packets since communication with AWS is sent over TLS connecting using MQTT or HTTPS but all I saw were UDP packets. I'm not sure if there's something I'm supposed to do in Wireshark to be able to see the traffic. 

Would I be able to enable MQTT_DEBUG in the embedded c?

Since I am only seeing UDP packets on Wireshark I'm not 100% sure the PINGREQ is going out but when I debug the code is getting to the keepalive portion and the send packet for the ping is returning SUCCESS.

scaprile, 2020-01-28:
Paho generates MQTT, that goes over TCP. If you are not able to see the traffic, to know if your device is working properly requires a wild guess. I suggest you disable TLS (if possible) or use Eclipse public broker or any other with a plain MQTT over TCP 1883 connection you can actually see to be sure your device works as expected. (you can also install your own Mosquitto or similar)
MQTT_DEBUG is just a #define. The only trick is retargetting [f]printf(), what you will learn probably from your compiler vendor. The other trick is that the C client (not the C++) does not seem to have this facility (according to grep). You didn't tell which one you use.
I don't get the UDP stuff, afaik that is TCP and you open the network connection before actually calling the client, so the transport should be what you decide it to be (server permitting...).

CIPop, 2023-07-19:
I have verified that both 
- v3 https://github.com/eclipse/paho.mqtt.embedded-c/blob/develop/MQTTPacket/samples/ping.c 
- v5 https://github.com/eclipse/paho.mqtt.embedded-c/blob/develop/MQTTPacket/samples/v5ping.c 
 
work with at least two different broker implementations but did not test with AWS.

icraggs, 2023-07-27:
I think this was answered.

closed 2023-07-27 by icraggs