eclipse-paho/paho.mqtt.embedded-c #207: QoS wairforcompletation Not existing!

Mauricio-Alencar, 2020-05-10:
is there any implementation for the MQTTClient_waitForCompletion () function in the mqtt c embedded client, as I need to work with QoS 2

scaprile, 2020-05-10:
Did you happen to search the other issues in order to find this answer to your question ?
https://github.com/eclipse/paho.mqtt.embedded-c/issues/201#issuecomment-578751712

Mauricio-Alencar, 2020-05-10:
So, I already researched, and even read all of these posts that you comment before opening this question here, or the problem in question is in some C do Paho customers there is the implementation of the following entries:

`  MQTTClient_deliveryToken token;`
   ` MQTTClient_publishMessage(client, topic, &pubmsg, &token); `
   ` MQTTClient_waitForCompletion(client, token, 1000L); `

While in Lib C Embbeded there is no token implementation with the Waitforcompletation function ... So I was wondering if anyone has already implemented it to be able to share there.