eclipse-paho/paho.mqtt.embedded-c #211: client disconnected after call unsbscribe

americaboy, 2020-07-16:
why my client will disconnect after I call unsbscribe?
in logs,I find "Error from MQTTAsync_cycle() - removing socket 69"，But I have no idea about it.
What causes it?

os: Android 8.1
logs as below:
7233-7233    =========================================================
7233-7233                       Trace Output
7233-7233    Product name: Eclipse Paho Asynchronous MQTT C Client Library
7233-7233    Version: 1.2.0
7233-7233    Build level: Thu Jul 16 15:30:18 CST 2020
7233-7233    OpenSSL version: BoringSSL
7233-7233    OpenSSL flags: BoringSSL
7233-7233    OpenSSL build timestamp: BoringSSL
7233-7233    OpenSSL platform: BoringSSL
7233-7233    OpenSSL directory: BoringSSL
7233-7233    /proc/version: Linux version 4.4.178-perf (hryt@hryt-Precision-5820-Tower) (gcc version 4.9.x 20150123 (prerelease) (GCC) ) #12 SMP PREEMPT Tue May 26 18:04:12 CST 2020

7233-7233    =========================================================

7233-7233    19700101 080000.000 (2582722984)  (1)> Socket_outInitialize:131
7233-7233    19700101 080000.000 (2582722984)   (2)> SocketBuffer_initialize:106
7233-7233    19700101 080000.000 (2582722984)   (2)< SocketBuffer_initialize:114 (0)
7233-7233    19700101 080000.000 (2582722984)  (1)< Socket_outInitialize:144
7233-7233    19700101 080000.000 (2582722984)  (1)> SSLSocket_initialize:447
7233-7233    19700101 080000.000 (2582722984)   (2)> SSL_create_mutex:346
7233-7233    19700101 080000.000 (2582722984)   (2)< SSL_create_mutex:352 (0)
7233-7233    19700101 080000.000 (2582722984)   (2)> SSL_create_mutex:346
7233-7233    19700101 080000.000 (2582722984)   (2)< SSL_create_mutex:352 (0)
7233-7233    20200715 002831.172 (2582722984)  (1)< SSLSocket_initialize:493 (1)
7233-7233    20200715 002831.172 (2582722984)  (1)> MQTTStrncpy:899
7233-7233    20200715 002831.172 (2582722984)  (1)< MQTTStrncpy:909
7233-7233    20200715 002831.172 (2582722984)  (1)> MQTTStrncpy:899
7233-7233    20200715 002831.172 (2582722984)  (1)< MQTTStrncpy:909
7233-7233    20200715 002831.172 (2582722984)  (1)> MQTTPersistence_create:52
7233-7233    20200715 002831.172 (2582722984)  (1)< MQTTPersistence_create:103 (0)
7233-7233    20200715 002831.172 (2582722984)  (1)> MQTTPersistence_initialize:118
7233-7233    20200715 002831.172 (2582722984)  (1)< MQTTPersistence_initialize:126 (0)
7233-7233    20200715 002831.172 (2582722984)  (1)> MQTTAsync_restoreCommands:1128
7233-7233    20200715 002831.178 0 commands restored for client DFTIDCM1234567892
7233-7233    20200715 002831.178 (2582722984)  (1)< MQTTAsync_restoreCommands:1166 (0)
7233-7233    20200715 002831.178 (2582722984)  (1)> MQTTPersistence_restoreMessageQueue:785
7233-7233    20200715 002831.178 0 queued messages restored for client DFTIDCM1234567892
7233-7233    20200715 002831.178 (2582722984)  (1)< MQTTPersistence_restoreMessageQueue:824 (0)
7233-7233    20200715 002831.178 (2582722984) (0)< MQTTAsync_createWithOptions:717 (0)
7233-7233    20200715 002831.178 (2582722984) (0)> MQTTAsync_setConnected:2789
7233-7233    20200715 002831.178 (2582722984) (0)< MQTTAsync_setConnected:2801 (0)
7233-7233    20200715 002831.178 (2582722984) (0)> MQTTAsync_setCallbacks:2675
7233-7233    20200715 002831.178 (2582722984) (0)< MQTTAsync_setCallbacks:2689 (0)
7233-7233    20200715 002831.178 (2582722984) (0)> MQTTAsync_connect:2991
7233-7233    20200715 002831.178 (2582722984)  (1)> Thread_start:69
7233-7233    20200715 002831.178 (2582722984)  (1)< Thread_start:79
7233-7233    20200715 002831.178 (2582722984)  (1)> Thread_start:69
7233-7233    20200715 002831.178 (2582722984)  (1)< Thread_start:79
7233-7233    20200715 002831.178 (2582722984)  (1)> MQTTAsync_addCommand:1176
7233-7233    20200715 002831.178 (2582722984)   (2)> Thread_signal_cond:403
7233-7233    20200715 002831.183 (2582722984)   (2)< Thread_signal_cond:408 (0)
7233-7233    20200715 002831.183 (2582722984)  (1)< MQTTAsync_addCommand:1210 (0)
7233-7233    20200715 002831.183 (2582722984) (0)< MQTTAsync_connect:3343 (0)
7233-7233    20200715 002831.183 (2582722984) (0)> MQTTProperties_free:382
7233-7233    20200715 002831.183 (2582722984) (0)< MQTTProperties_free:405
7233-7233    20200715 002831.183 (2582722984) (0)> MQTTProperties_free:382
7233-7233    20200715 002831.183 (2582722984) (0)< MQTTProperties_free:405
7233-7264    20200715 002831.183 (2509497584) (0)> MQTTAsync_receiveThread:2297
7233-7264    20200715 002831.183 (2509497584)  (1)> MQTTAsync_cycle:4045
7233-7264    20200715 002831.183 (2509497584)   (2)> Socket_getReadySocket:249
7233-7264    20200715 002831.183 (2509497584)   (2)< Socket_getReadySocket:321 (0)
7233-7264    20200715 002831.183 (2509497584)   (2)> MQTTAsync_sleep:478
7233-7263    20200715 002831.183 (2510533872) (0)> MQTTAsync_sendThread:2013
7233-7263    20200715 002831.183 (2510533872)  (1)> MQTTAsync_processCommand:1530
[... 395 lines cut ...]
7233-7263    20200715 002832.392 (2510533872)          (9)< Socket_writev:490 (2)
7233-7263    20200715 002832.392 (2510533872)         (8)< Socket_putdatas:567 (0)
7233-7263    20200715 002832.392 (2510533872)        (7)< WebSocket_putdatas:934 (0)
7233-7263    20200715 002832.393 (2510533872)       (6)< MQTTPacket_send:222 (0)
**7233-7263    20200715 002832.393 69 DFTIDCM1234567892 -> DISCONNECT (0)**
7233-7263    20200715 002832.393 (2510533872)      (5)< MQTTPacket_send_disconnect:528 (0)
7233-7263    20200715 002832.393 (2510533872)      (5)> WebSocket_close:510
7233-7263    20200715 002832.393 (2510533872)      (5)< WebSocket_close:560
7233-7263    20200715 002832.393 (2510533872)      (5)> SSLSocket_close:918
7233-7263    20200715 002832.393 (2510533872)       (6)> SSLSocket_destroyContext:905
7233-7263    20200715 002832.393 (2510533872)       (6)< SSLSocket_destroyContext:909
7233-7263    20200715 002832.393 (2510533872)      (5)< SSLSocket_close:930 (1)
7233-7263    20200715 002832.393 (2510533872)      (5)> Socket_close:630
7233-7263    20200715 002832.393 (2510533872)       (6)> Socket_close_only:604
7233-7263    20200715 002832.393 (2510533872)       (6)< Socket_close_only:618 (0)
7233-7263    20200715 002832.393 (2510533872)       (6)> Socket_abortWrite:953
7233-7263    20200715 002832.393 (2510533872)       (6)< Socket_abortWrite:971 (0)
7233-7263    20200715 002832.393 (2510533872)       (6)> SocketBuffer_cleanup:153
7233-7263    20200715 002832.393 (2510533872)       (6)< SocketBuffer_cleanup:165
7233-7263    20200715 002832.393 Removed socket 69
7233-7263    20200715 002832.393 Reset max fdp1 to 1
7233-7263    20200715 002832.393 (2510533872)      (5)< Socket_close:657
7233-7263    20200715 002832.393 (2510533872)     (4)< MQTTAsync_closeOnly:2830
7233-7263    20200715 002832.393 (2510533872)    (3)< MQTTAsync_closeSession:2843
7233-7263    20200715 002832.393 Calling connectionLost for client DFTIDCM1234567892

scaprile, 2020-07-16:
Perhaps the proper repo is [this one](https://github.com/eclipse/paho.mqtt.c) ?
Explain what you are calling and how you are calling it.

americaboy, 2020-07-20:
**first,I call client->connect(connOpts, nullptr, *callback) to establish connection with broker.***

```
// connect broker
    try {
        LOGD(TAG, "client connect broker");
        client->connect(connOpts, nullptr, *cb);
        LOGD(TAG, "client connect server done");
    } catch(const mqtt::exception& exc) {
        LOGD(TAG, "connect server exception:%s", exc.what());
    }
```
**next,I called subscribe(topic, QOS_2) in callback's on_sucess{} or connected{}.**
```
class callback : public virtual mqtt::callback,
                 public virtual mqtt::iaction_listener

{
    // Counter for the number of connection retries
    int nretry_;
    // The MQTT client
    mqtt::async_client& cli_;
    // Options to use if we need to reconnect
    mqtt::connect_options& connOpts_;
    // An action listener to display the result of actions.
    action_listener subListener_;

    MyService* service;

    void reconnect() {
        try {
            std::this_thread::sleep_for(std::chrono::milliseconds(1000));
            LOGD(TAG, "client Connection reconnect");
            cli_.connect(connOpts_, nullptr, *this);
        } catch(const mqtt::exception& exc) {
            LOGD(TAG, "client reconnect exception:%s", exc.what());
        }
    }

    // Re-connection failure
    void on_failure(const mqtt::token& tok) override {
        LOGD(TAG, "client Connection failed");
        reconnect();
    }

    // Re-connection success
    void on_success(const mqtt::token& tok) override {
        LOGD(TAG, "client Connection success");
    }

    void connected(const std::string& cause) override {
        LOGD(TAG, "client Connection connected");
        if(service != nullptr) {
            string topic1 = service->getTopic();
            if(!topic1.empty()) {
                try {
                    LOGD(TAG, "subscribe topic1:%s", topic1.c_str());
                    cli_.subscribe(topic1, QOS_2);
                } catch(const mqtt::exception& exc) {
                    LOGD(TAG, "subscribe topic1 failed:%s", exc.what());
                }
            } 
            string topic2 = service->getTopic();
            if(!topic2.empty()) {
                try {
                    LOGD(TAG, "subscribe topic2 topic:%s", topic2.c_str());
                    cli_.subscribe(topic2, QOS_2);
                } catch(const mqtt::exception& exc) {
                    LOGD(TAG, "subscribe topic2 failed:%s", exc.what());
                }
            }
        }
    }

    // Callback for when the connection is lost.
    // This will initiate the attempt to manually reconnect.
    void connection_lost(const std::string& cause) override {
        LOGD(TAG, "client Connection lost");
        reconnect();
    }

    // Callback for when a message arrives.
    void message_arrived(mqtt::const_message_ptr msg) override {
        LOGD(TAG, "Message arrived topic:%s data:%s", msg->get_topic().c_str(),
             msg->to_string().c_str());
    }

    void delivery_complete(mqtt::delivery_token_ptr token) override {
        LOGD(TAG, "delivery_complete...");
    }

public:
    callback(mqtt::async_client& cli, mqtt::connect_options& connOpts, MyService* service)
        : nretry_(0)
        , cli_(cli)
        , connOpts_(connOpts)
        , subListener_("Subscription")
        , service(service) {
    }
};
```
**then I called unsubscribe(topic) when I recieve a notification of user changed from os.**
```
void MyService::onCurrentUserChange(const std::string& userId) {
    LOGD(TAG, "new user=%s", userId.c_str());
    string last_userId = current_userId;
    string last_topic = current_topic;

    // user was changed
    if(!userId.empty() && userId != last_userId) {
        current_userId = userId;
        current_topic = getTopic(current_userId);

        // unsubscribe old user topic
        if(client != nullptr && client->is_connected() && !last_topic.empty()) {
            try {
                LOGD(TAG, "unsubscribe old topic:%s", last_topic.c_str());
                client->unsubscribe(last_topic);
            } catch(const mqtt::exception& exc) {
                LOGD(TAG, "unsubscribe  old topic failed:%s", exc.what());
            }
        }
        // subscribe new  user topic
        if(client != nullptr && client->is_connected()) {
            try {
                LOGD(TAG, "subscribe new topic:%s", current_topic.c_str());
                client->subscribe(current_topic, DEFAULT_QOS);
            } catch(const mqtt::exception& exc) {
                LOGD(TAG, "subscribe new topic failed:%s", exc.what());
            }
        }
    }
}
```

scaprile, 2020-07-20:
You're at the wrong repo.
There is no reference for 'async_client' at the embedded-c repo.
The class definition for your init call seems to be [here](https://github.com/eclipse/paho.mqtt.cpp/blob/master/src/mqtt/async_client.h), so I bet the proper repo is perhaps [this one](https://github.com/eclipse/paho.mqtt.cpp/). Try there,

americaboy, 2020-07-21:
sorry for that I picked the wrong repo!
thx!

closed 2023-11-29 by icraggs

Cartube12, 2024-12-19:
U find another way to find?