eclipse-paho/paho.mqtt.embedded-c #213: Timer/Network is undeclared when using MQTTClient.h

kiranpradeep, 2020-11-02:
I felt the below code should still compile. 

# Code

```console
paho@mqtt:embedded-c$ cat main.c
#include "MQTTClient.h"
int main() {
  return 0;
}
```

# Console output
```console
paho@mqtt:embedded-c$ gcc -c main.c -I MQTTPacket\src -I MQTTClient-C\src
```

> In file included from main.c:1:
> MQTTClient-C/src/MQTTClient.h:70:23: error: unknown type name ‘Timer’
>    70 | extern void TimerInit(Timer*);
>       |                       ^~~~~
> MQTTClient-C/src/MQTTClient.h:71:28: error: unknown type name ‘Timer’
>    71 | extern char TimerIsExpired(Timer*);
>       |                            ^~~~~
> MQTTClient-C/src/MQTTClient.h:72:30: error: unknown type name ‘Timer’
>    72 | extern void TimerCountdownMS(Timer*, unsigned int);
>       |                              ^~~~~
> MQTTClient-C/src/MQTTClient.h:73:28: error: unknown type name ‘Timer’
>    73 | extern void TimerCountdown(Timer*, unsigned int);
>       |                            ^~~~~
> MQTTClient-C/src/MQTTClient.h:74:24: error: unknown type name ‘Timer’
>    74 | extern int TimerLeftMS(Timer*);
>       |                        ^~~~~
> MQTTClient-C/src/MQTTClient.h:126:5: error: unknown type name ‘Network’
>   126 |     Network* ipstack;
>       |     ^~~~~~~
> MQTTClient-C/src/MQTTClient.h:127:5: error: unknown type name ‘Timer’
>   127 |     Timer last_sent, last_received;
>       |     ^~~~~
> MQTTClient-C/src/MQTTClient.h:144:51: error: unknown type name ‘Network’
>   144 | DLLExport void MQTTClientInit(MQTTClient* client, Network* network, unsigned int command_timeout_ms,

scaprile, 2020-11-02:
You are not following the build procedure and you are facing some dependencies that are not defined in the include files you are including.
See the samples, see if there is one for your OS, or learn from the closest one.

kiranpradeep, 2020-11-02:
The build runs perfectly on Ubuntu 20.04 for which I was testing. I reported this only since, in **any** other project, I have not seen a header file whose inclusion itself would result in a compiler error. After the above reply, I looked again and saw a `MQTTCLIENT_PLATFORM_HEADER` which is used by the build. Apologies, I should have looked at this earlier, before raising this issue.

closed 2020-11-02 by kiranpradeep